tuanpmt/esp_mqtt #5: Makefile etc

aidanruff, 2015-01-01:
So, I've bodged the makefile from your previous incarnation of the MQTT software and I can now successfully compile and re-compile it.

However, once it's logged onto my MQTT server and subscribed to the test messages it just sits there doing nothing!

If I send an MQTT message to one of it's topics, it doesn;t respond and if I send any serial data it doesn't respond either.

I think that this has great potential though. 

I've thought a lot about it and when you get it further developed it would be great to have a consistent set of messages:- something like

Send
MQTT:<topic>:<message text><CR>
response 
OK<CR>
or
<FAIL>
or
SUBSCRIBE:<topic>
response
OK<CR>
or
FAIL<CR>

When the wifi has connected and the MQTT server has connected maybe a simple message like
READY<CR>

I was thinking about how you might connect a newly installed ESP module and perhaps if GPIO2 is held to ground at power up, you could generate a WiFi AP with the SSID as the MAC number of the module.

A simple config web page would then give the following:-
AP SSID and password.
MQTT server URL and port number
MQTT server user and password
An outgoing data topic if required
An incoming data topic if required
Display the units MAC number

This would allow anyone to connect to a new unit with a phone or tablet to configure it.

I can see applications for this in home automation. My friend Peter Scargill and I are designing home automation devices and we've been using NRF24L01 radio modules and ATMEL RF processors with a lightweight mesh. 

However, I think that ESP modules with MQTT software is far more useful.

I'd be happy to give you a few mains control units to play with when we get the PCBs finished and integrated with your MQTT software

Regards
Aidan

PS This is the current version of my mains switching unit with a 10 amp relay and a DS1802 temperature sensor

[image]

tuanpmt, 2015-01-03:
hello Aidan
This library design for native ESP8266 SoC, used for an application run stand alone in this SoC, not for serial. Of course, you can use this library for your application run with other MCU and communicate via serial port.
I have done other library for Web configuration, and can be integrated with this, but i want to make this library clearly

aidanruff, 2015-01-04:
Ah ok! Many thanks for th info
Regards
Aidan

closed 2015-01-04 by tuanpmt