tuanpmt/esp_mqtt #11: no DVES_HOME found??

scargill, 2015-01-04:
Just downloaded the latest version - I note that things have been moved around.

In the user_config.file I entered my STA_SSID and STA_PASS for my router - and the name of my MQTT host and the port.  

I compiled for ALL then FLASH - and this is what I'm getting...

no DVES_HOME found, reconnect after 1s.....
Station_NO_AP_found... etc

??

tuanpmt, 2015-01-04:
hello scargill,
Did you tried to change value of CFG_HOLDER when you change default configuration
```c
if(sysCfg.cfg_holder != CFG_HOLDER){
	sysCfg.cfg_holder = CFG_HOLDER;
	........
}
```

It means: if the value of the cfg_holder != CFG_HOLDER then assign value of CFG_HOLDER to cfg_holder, reload all default values, and save
**When you flash ESP, section to save configuration will not be erased**

scargill, 2015-01-04:
Yes, absolutely - I realised that when I read the README.. here's a fuller output - you'll see it connected to my router and even subscribed...

DNS: found ip 109.170.132.114
TCP: connecting...
TCP: Reconnect to home.scargill.org:1883
TCP: Connect to domain home.scargill.org:1883
TCP:Reconect to: home.scargill.org:1883
DNS: found ip 109.170.132.114
TCP: connecting...
TCP: Reconnect to home.scargill.org:1883
TCP: Connect to domain home.scargill.org:1883
TCP:Reconect to: home.scargill.org:1883
DNS: found ip 109.170.132.114
TCP: connecting...
TCP: Reconnect to home.scargill.org:1883

It's doing the reconnect continuously - every second...

scargill, 2015-01-04:
Yes - I realised when I failed the first time that you were using a checksum in the FLASH to store defaults..

Here's a longer output. As you can see it connects to the router (all of this was/is working flawlessly on the version before you added SSL)....

It's just sitting a loop repeating every seconds or so.

load ...
MQTT_InitConnection
MQTT_InitClient
WIFI_INIT

System started ...
mode : sta(18:fe:34:9e:d7:6c)
add if0
scandone
STATION_IDLE
add 0
aid 2
pm open phy_2,type:2 0 0
cnt 

connected with loft-east, channel 6
dhcp client start...
STATION_IDLE
ip:192.168.0.16,mask:255.255.255.0,gw:192.168.0.138
TCP: Connect to domain home.scargill.org:1883
MQTT: subscribe, topic"/test/topic" at broker home.scargill.org:1883
MQTT: subscribe, topic"/test2/topic" at broker home.scargill.org:1883
DNS: found ip 109.170.132.114
TCP: connecting...
client handshake start.
TCP: Reconnect to home.scargill.org:1883
TCP: Connect to domain home.scargill.org:1883
TCP:Reconect to: home.scargill.org:1883
DNS: found ip 109.170.132.114
TCP: connecting...
TCP: Reconnect to home.scargill.org:1883
TCP: Connect to domain home.scargill.org:1883
TCP:Reconect to: home.scargill.org:1883
DNS: found ip 109.170.132.114
TCP: connecting...
TCP: Reconnect to home.scargill.org:1883
TCP: Connect to domain home.scargill.org:1883
TCP:Reconect to: home.scargill.org:1883
DNS: found ip 109.170.132.114

tuanpmt, 2015-01-04:
If you don't use SSL broker, please change security mode to SEC_NONSSL: 
```c
MQTT_InitConnection(&mqttClient, sysCfg.mqtt_host, sysCfg.mqtt_port, SEC_NONSSL);
```

scargill, 2015-01-04:
YES you are right - the change to non-secure works - thank you.

However, I sent a message to /test/topic

The message was "hmm"

The reply as you see in the serial... is this..

MQTT topic: /test/topichmm, data: hmm 

As you can see the data is attached to the end of the topic as well as being on it's own.

The only place I can see where you do this is in user_main.c

void mqttDataCb(uint32_t *args, const char* topic, uint32_t topic_len, const char *data, uint32_t lengh)
{
	INFO("MQTT topic: %s, data: %s \r\n", topic, data);
}

and to be honest apart from "lengh" being spelled incorrectly (length I assume) I cannot see why this is happening.

scargill, 2015-01-04:
If when subscribing we want access to the unique number you used originally instead of /test/ - how do we access that number?

void wifiConnectCb(uint8_t status)
{
	if(status == STATION_GOT_IP){
		MQTT_Connect(&mqttClient);

		MQTT_Subscribe(&mqttClient, "/test/topic");
		MQTT_Subscribe(&mqttClient, "/test2/topic");
	}
}

Would this still work..

os_sprintf(myarray, "/%08X/data", sysCfg.device_id);

and hence 

MQTT_Subscribe(&mqttClient, myarray);

tuanpmt, 2015-01-04:
Please see new Readme update tuanpmt@3ca9908267d7ecd296dad9ed5ec3a1641e0bfbdf

scargill, 2015-01-04:
ALMOST!   You're now missing the topic altogether :-)

MQTT topic: , data: VERY NICE 
MQTT: sending publish...

scargill, 2015-01-04:
Simple wrong buffer- change first use of databuf to topicbuf, sorted...

void mqttDataCb(uint32_t *args, const char* topic, uint32_t topic_len, const char *data, uint32_t data_len)
{
	char topicBuf[64], dataBuf[64];
	MQTT_Client* client = (MQTT_Client*)args;

	os_memcpy(topicBuf, topic, topic_len);
	topicBuf[topic_len] = 0;

	os_memcpy(dataBuf, data, data_len);
	dataBuf[data_len] = 0;

	INFO("MQTT topic: %s, data: %s \r\n", topicBuf, dataBuf);

	/* Echo back to /echo channel*/
	MQTT_Publish(client, "/echo", dataBuf, data_len, 0, 0);
}

commit f837152b (2015-01-04): fixed #11, silly mistake
closed 2015-01-04 by tuanpmt