tuanpmt/esp_mqtt #12: Safe to publish callback?

betgear, 2015-01-04:
Hi,
Thanks for adding the callback for published data, which brings forward another issue.

The code I'm trying to implement is to send a temperature reading, then sleep for 15 minutes and repeat. The MQTT_Publish command will ignore any messages until the system is in a ready state (connState == MQTT_DATA) to be able to send messages.

Therefore I need to determine when it is safe to send a message before going to sleep. Looking at the code, the last callback appears to be the mqttConnectCb function. I tried adding my code to read the temperature sensor at the end of this callback, but the MQTT_Publish is ignored (I'm presuming the state is incorrect to send the message).

Is there a way / technique within the callback system to know when i can send a message?

Can the messages be queued to be sent?

Would another callback for readyToSend be practical?

tuanpmt, 2015-01-04:
now in mqttConnectCb function, you can publish a message

betgear, 2015-01-04:
This appears to be broken again with the latest version :(

All I've done is insert

MQTT_Publish(&mqttClient, "/test/topic", "Hello World 99", 14, 2, 0);

at the end of the mqttConnectedCb, and the message never gets sent.

ip:192.168.1.47,mask:255.255.255.0,gw:192.168.1.1
TCP: Connect to ip  192.168.1.147:1883
MQTT: Connected to broker 192.168.1.147:1883
MQTT: Send mqtt connection info, to broker 192.168.1.147:1883
TCP: Sent
TCP: data received
MQTT: Connected to 192.168.1.147:1883
MQTT: Connected
MQTT: subscribe, topic"/test/topic" at broker 192.168.1.147:1883
MQTT: subscribe, topic"/test2/topic" at broker 192.168.1.147:1883
TCP: Sent
TCP: data received
MQTT: Subscribe successful
TCP: Sent
TCP: data received
MQTT: Subscribe successful

MQTT: Send keepalive packet to 192.168.1.147:1883!
TCP: Sent
MQTT: Published
TCP: data received

betgear, 2015-01-10:
Any news on this, my project is on hold until this is fixed.

Thanks.

commit 34a3ea49 (2015-01-04): Call published event after subscribe, resolved #12
commit 1995cb81 (2015-01-13): Change subscribe queue, add publish queue, resolved #24, fixes #12
closed 2015-01-13 by tuanpmt