tuanpmt/esp_mqtt #14: Compile error on latest version ...

Werquinx, 2015-01-04:
Cf:

CC user/mqtt.c
user/mqtt.c: In function 'mqtt_tcpclient_recv':
user/mqtt.c:224:8: error: passing argument 1 of 'client->publishedCb' from incompatible pointer type [-Werror]
        client->publishedCb(client);
        ^
user/mqtt.c:224:8: note: expected 'uint32_t *' but argument is of type 'struct MQTT_Client *'
user/mqtt.c:238:8: error: passing argument 1 of 'client->publishedCb' from incompatible pointer type [-Werror]
        client->publishedCb(client);
        ^
user/mqtt.c:238:8: note: expected 'uint32_t *' but argument is of type 'struct MQTT_Client *'
user/mqtt.c: In function 'mqtt_tcpclient_sent_cb':
user/mqtt.c:298:4: error: passing argument 1 of 'client->publishedCb' from incompatible pointer type [-Werror]
    client->publishedCb(client);
    ^
user/mqtt.c:298:4: note: expected 'uint32_t *' but argument is of type 'struct MQTT_Client *'
cc1.exe: all warnings being treated as errors
C:/Users/Xavier/git/esp_mqtt/Makefile:160: recipe for target 'build/user/mqtt.o' failed
mingw32-make.exe: *** [build/user/mqtt.o] Error 1

13:15:16 Build Finished (took 969ms)

tuanpmt, 2015-01-04:
checkout again

Werquinx, 2015-01-04:
Changed all offending lines to : 

		if(client->publishedCb)
			client->publishedCb((uint32_t*)client);

Now compiles fine and works OK ...

closed 2015-01-04 by tuanpmt