tuanpmt/esp_mqtt #17: wrong compiler options

struhalla, 2015-01-06:
I think the options for compiling release and debug versions are wrong, instead of

ifeq ($(FLAVOR),debug)
CFLAGS += -g -O2
LDFLAGS += -g -O2
endif
ifeq ($(FLAVOR),release)
CFLAGS += -g -O0
LDFLAGS += -g -O0
endif

it should be

ifeq ($(FLAVOR),debug)
CFLAGS += -g -O0
LDFLAGS += -g -O0
endif
ifeq ($(FLAVOR),release)
CFLAGS += -g -O2
LDFLAGS += -g -O2
endif

tuanpmt, 2015-01-06:
-O0 : Do not optimize. 
-O2 : Optimize even more. 

https://gcc.gnu.org/onlinedocs/gcc-3.2/gcc/Optimize-Options.html#Optimize%20Options

struhalla, 2015-01-06:
without changing the makefile it produces a release version with min optimization (-O0) and a debug version with max optimization (-O2)

commit 34e3cb70 (2015-01-06): correct the Makefile, fixes #17
closed 2015-01-06 by tuanpmt