tuanpmt/esp_mqtt #19: Reset on message with empty data

RickP, 2015-01-07:
If I publish a message with empty data to a subscribed channel the device resets. I'm using the example code with SSL auth switched off.

    Fatal exception (28):
    epc1=0x4000df7c, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
    rm match
    pm close 7 0 0/134597350

     ets Jan  8 2013,rst cause:4, boot mode:(3,7)

kennpie, 2015-01-10:
Just a note to say I have the exact same issue. I'm also using no SSL, connecting to Mosquitto.

I'm sure it's a simple fix, and I may go looking for the issue myself, but my github skills aren't yet good enough to forward a fix if I find one.

(And thanks alot for this code!)

tuanpmt, 2015-01-10:
Please comment here, i will add yor fixes. Thanks

scargill, 2015-01-16:
Tunanmpt - your efforts are MOST greatly appreciated...

closed 2015-01-17 by tuanpmt

scargill, 2015-01-17:
What a good man - oh, well here goes another round of downloads :-)

commit 63eff771 (2015-01-17): fixed #19 Reset on message with empty data