tuanpmt/esp_mqtt #22: Last Will LWT

jcwoltz, 2015-01-10:
Does this support LWT? If so can you show an example? 

If not, can you please add it? 

Thank you

tuanpmt, 2015-01-14:
Hello,
This adrely support LWT:
you can setup in *MQTT_InitClient* file ***mqtt.c***
```c
char willTopic[] = "/lwt";
char willMessage[] = "offline";
mqttClient->connect_info.will_topic = willTopic;
mqttClient->connect_info.will_message = willMessage;
mqttClient->will_qos = 0;
mqttClient->will_retain = 0;
```

closed 2015-01-14 by tuanpmt

jcwoltz, 2015-01-14:
When I do that and try to compile, I get:
CC user/mqtt.c
user/mqtt.c: In function 'MQTT_InitClient':
user/mqtt.c:486:12: error: 'MQTT_Client' has no member named 'will_qos'
  mqttClient->will_qos = 0;
            ^
user/mqtt.c:487:12: error: 'MQTT_Client' has no member named 'will_retain'
  mqttClient->will_retain = 0;
            ^

tuanpmt, 2015-01-14:
Sorrry, mistaken
```c
mqttClient->connect_info.will_qos = 0;
mqttClient->connect_info.will_retain = 0;
```

jcwoltz, 2015-01-18:
Thank you for adding a better way to support LW&T now.

commit 4f4209c0 (2015-01-14): add example for support Last Will and Testament LWT, resolve #22