tuanpmt/esp_mqtt #24: Queue not working

scargill, 2015-01-12:
HI

Not sure if there is supposed to be a publish queue but if not is there a chance you could fix this please... sometimes I need to be able to send 2 messages, one after another - and I can't as only the last one gets through as if the first is being over-written. Would REALLY appreciate this?

scargill, 2015-01-12:
I just thought of another reason we need the queue working - one ESP8266 might have several functions - I want to send a temperature message to openHab and also a humidity message -  I can't send them both in the same message - so I have to send 2 - which right now means staggering them and that doesn't work well if I've had a request from the syttem for that data - we could of course implement this in a timer but seems so daft if it's supposed to be there? Is the queue supposed to work?

scargill, 2015-01-12:
Cannot get mqttPublishedCb to fire at end of a publication

betgear, 2015-01-13:
Hi Peter,

Get the latest version of the code, I believe it's fixed in that.

Cheers

Glen.

scargill, 2015-01-13:
Indeed it is and I’ve implemented a working queue until the real one comes along – together with RTC. More at http://scargill.wordpress.com

scargill, 2015-01-13:
Oh excellent.

 

Closed #24 <https://github.com/tuanpmt/esp_mqtt/issues/24>  via  <https://github.com/tuanpmt/esp_mqtt/commit/1995cb812a54f001362661f30281e31e3e6c9708> 1995cb8.

scargill, 2015-01-13:
Sent time to 3 publications one after another – and the software crashed. First time I’ve seen it crash.

mhwlng, 2015-01-13:
I pusblish one string, every minute.
now the data for the first publish is re-used every time, instead of using the new data.

closed 2015-01-14 by tuanpmt

scargill, 2015-01-16:
Hi

 

I just wasted ages thinking your new code was broken – then tried my own queue and that broke to – after a lot of wasted time.. it hit me…

 

If the ESP8266 is subscribed to a topic – and someone publishes the topic with NO MESSAGE (I’m using MQTT-SPY) the program crashes.  I think it would be well worthwhile trapping this and making sure it treats this as an empty message.

 

With data in the message

 

TCP: data received

MQTT topic: myleds, data: {0} 

MQTT: sending publish...

TCP: Sent

MQTT: Published

 

With an empty message

 

TCP: data received

Fatal exception (28):

epc1=0x4000df7c, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

rm match

pm close 7 0 0/163406647

 

ets Jan  8 2013,rst cause:4, boot mode:(3,7)

 

wdt reset

load 0x40100000, len 31624, room 16 

tail 8

chksum 0xec

load 0x3ffe8000, len 2652, room 0 

tail 12

chksum 0xfa

ho 0 tail 12 room 4

load 0x3ffe8a60, len 7408, room 12 

tail 4

chksum 0x4c

 

Closed #24 <https://github.com/tuanpmt/esp_mqtt/issues/24>  via  <https://github.com/tuanpmt/esp_mqtt/commit/008fd11b590f671c5093ef73dd9b08f4eb488aa1> 008fd11.

tuanpmt, 2015-01-16:
I've checked very carefully, it still worked fine. Have you checkout the latest code version yet?

scargill, 2015-01-16:
I downloaded a version yesterday and I am using that. But exactly the same happens with much earlier versions.

 

I can do this 100% repeatedly – see my code above… your code – I’ve added my stuff in. 

 

So I have 4 topics – I’ve added some real time clock (1 second interrupt) – I check for “time” every 5 seconds and update a single LONG in the interrupt… temperature sensor (does not matter if it is not plugged into GPIO2) – LEDS are GPIO0 – again does not matter if not plugged in.

 

The point being – it is utterly reliable as long as there is something send in the message from MQTT-SPY – but EVERY time I leave the message empty – it crashes – both my original version where I wrote the queue and your new one.

 

In the code you will easily see my added code – for temperature sensor (DHT22) – real time clock (a simple LONG incremented every second – and code to fill a struct with time and date) and some port IO (feel free to use anything – but warning the WS2812B code does not work due to compiler issues- need someone to write a little bit of it in assembler).

 

 <https://code.google.com/p/mqtt-spy/> https://code.google.com/p/mqtt-spy/

 

/* main.c -- MQTT client example
*
* Copyright (c) 2014-2015, Tuan PM <tuanpm at live dot com>
* All rights reserved.
*
* Redistribution and use in source and binary forms, with or without
* modification, are permitted provided that the following conditions are met:
*
* * Redistributions of source code must retain the above copyright notice,
* this list of conditions and the following disclaimer.
* * Redistributions in binary form must reproduce the above copyright
* notice, this list of conditions and the following disclaimer in the
* documentation and/or other materials provided with the distribution.
* * Neither the name of Redis nor the names of its contributors may be used
* to endorse or promote products derived from this software without
* specific prior written permission.
*
* THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
* AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
* IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
* ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT OWNER OR CONTRIBUTORS BE
* LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
* CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF
* SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
* INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN
* CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
* ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
* POSSIBILITY OF SUCH DAMAGE.
*/
#include "ets_sys.h"
#include "driver/uart.h"
#include "osapi.h"
#include "mqtt.h"
#include "wifi.h"
#include "config.h"
#include "debug.h"
#include "gpio.h"
#include "user_interface.h"

#define WSGPIO 0

#define LED_GPIO 2
#define LED_GPIO_MUX PERIPHS_IO_MUX_GPIO2_U
#define LED_GPIO_FUNC FUNC_GPIO2
#define DELAY 5000 /* milliseconds for temperature 5 second intervals */

[... 302 lines cut ...]
	// os_timer_disarm(ETSTimer *ptimer)
	os_timer_disarm(&temperature_timer);
	// os_timer_setfn(ETSTimer *ptimer, ETSTimerFunc *pfunction, void *parg)
	os_timer_setfn(&temperature_timer, (os_timer_func_t *)temperature_cb, (void *)0);
	// void os_timer_arm(ETSTimer *ptimer,uint32_t milliseconds, bool repeat_flag)
	os_timer_arm(&temperature_timer, DELAY, 1);

	os_timer_disarm(&rtc_timer);
	os_timer_setfn(&rtc_timer, (os_timer_func_t *)rtc_cb, (void *)0);
	os_timer_arm(&rtc_timer, 1000, 1);

//	MQTT_InitConnection(&mqttClient, sysCfg.mqtt_host, sysCfg.mqtt_port, sysCfg.security);
	MQTT_InitConnection(&mqttClient, sysCfg.mqtt_host, sysCfg.mqtt_port, SEC_NONSSL);

	MQTT_InitClient(&mqttClient, sysCfg.device_id, sysCfg.mqtt_user, sysCfg.mqtt_pass, sysCfg.mqtt_keepalive);
	MQTT_OnConnected(&mqttClient, mqttConnectedCb);
	MQTT_OnDisconnected(&mqttClient, mqttDisconnectedCb);
	MQTT_OnPublished(&mqttClient, mqttPublishedCb);
	MQTT_OnData(&mqttClient, mqttDataCb);

	WIFI_Connect(sysCfg.sta_ssid, sysCfg.sta_pwd, wifiConnectCb);

	INFO("\r\nSystem started ...\r\n");
}
=

tuanpmt, 2015-01-16:
Ok, problem is empty data, but queue is working rightnow ?

scargill, 2015-01-16:
Oh yes, queue is working perfectly – but BOTH the version BEFORE the queue and AFTER the queue fail with empty data…  must be worth a quick fix?

tuanpmt, 2015-01-16:
Yup,
That is another issue opening #19

commit 1995cb81 (2015-01-13): Change subscribe queue, add publish queue, resolved #24, fixes #12
commit 008fd11b (2015-01-14): fixes #24, resolved #25