tuanpmt/esp_mqtt #27: Does not process data from second subscription immediately after connect

jcwoltz, 2015-01-17:
Hello,

When I subscribe to two topics, as in the initial example, during the mqttConnectedCb the data is not processed from the second subscription in the mqttDataCb. If I send data later, it does work. However, on boot and reconnect, it is not getting this second message. 

Similarly, If I subscribe to a wildcard, I do not get the topics passed to the mqttDataCb during the connected stage. It only gets the first message. 

I do not know if something else needs added to a queue to get this to work. The reason it is important to get the data from the 2 subscriptions is to restore the state on bootup or reconnect. In this test, I subscribe to two topics:
sensor/node1/control/led1
sensor/node1/control/led2

Below is the relevant output from this test. Thank you for your work. 

MQTT: Starting subscription
MQTT: subscribe, topic"sensor/node1/control/led1" at broker 104.200.29.35:1883
MQTT: subscribe, topic"sensor/node1/control/led2" at broker 104.200.29.35:1883
MQTT: queuing publish, length: 24...
MQTT: Sending..
TCP: Sent
MQTT: Published
TCP: data received
MQTT: Sending..
TCP: Sent
MQTT: Published
TCP: data received
MQTT topic: sensor/node1/control/led1, data: on
LED1 sub
LED1 sub data high
MQTT: Sending..
TCP: Sent
MQTT: Published
TCP: data received

tuanpmt, 2015-01-17:
can you show your test code?

jcwoltz, 2015-01-17:
sure, it is all at:
https://github.com/jcwoltz/esp_mqtt

jcwoltz, 2015-01-17:
I have pulled in you latest changes on mqtt.c/h into my setup.
I have also tested this with a fresh checkout of your repo, only changing my settings in user_config.h and changing the topic subscription in user_main.c

TCP: Connect to ip  104.200.29.35:1883
MQTT: Connected to broker 104.200.29.35:1883
MQTT: Sending..type: 1,qos: 4
TCP: Sent
TCP: data received
MQTT: Connected to 104.200.29.35:1883
MQTT: Connected
MQTT: Starting subscription
MQTT: subscribe, topic"sensor/node1/control/led1" at broker 104.200.29.35:1883
MQTT: subscribe, topic"sensor/node1/control/led2" at broker 104.200.29.35:1883
MQTT: queuing publish, length: 24...
MQTT: Sending..type: 8,qos: 4
TCP: Sent
TCP: data received
MQTT: Sending..type: 8,qos: 4
TCP: Sent
TCP: data received
deliver_publish
MQTT topic: sensor/node1/control/led1, data: on
LED1 sub
LED1 sub data high
MQTT: Sending..type: 3,qos: 0
TCP: Sent
MQTT: Published
TCP: data received

tuanpmt, 2015-01-17:
I don't understand your problem, "during the mqttConnectedCb the data is not processed from the second subscription in the mqttDataCb". If you want to process data from other mqtt client, if you need subscribe first. Or, publish a message with retain = 1

jcwoltz, 2015-01-17:
I have messages published with retain=1 at the following locations:
sensor/node1/control/led1
sensor/node1/control/led2

both locations have a msg of either on or off

If you look at the previous captures, led2 is never processed during bootup. If I send a msg after the node is up and running, it works fine. Does that make sense? Can we communicate some other way? something like IRC?

jcwoltz, 2015-01-17:
you can see my current user_main.c at:
https://github.com/jcwoltz/esp_mqtt/blob/master/user/user_main.c

I am subscribing to 2 topics. 
I only get data from the first topic at boot. Does that make sense?

tuanpmt, 2015-01-17:
Ok jcwolt,
I understood. I'll check tomorrow. It's 10PM now.

jcwoltz, 2015-01-17:
ok, thank you.

tuanpmt, 2015-01-18:
I have found and fixed the problem, the reason is because function espconn_sent send continuous data without the last check was sent successfully. I'll test the fixes and commit soon.

closed 2015-01-18 by tuanpmt

jcwoltz, 2015-01-18:
Thank you. This works great now.

commit 47bb4ad2 (2015-01-18): resolve #27, allocate memory for init data, add lwt api