tuanpmt/esp_mqtt #30: CFG_LOAD issues

theater, 2015-01-31:
Hi Tuan,
I have updated to your latest version with the makefile that clears out the configuration fields but I still have the issue with configuration not working correct. When I commented out the whole CFG load from memory I've got the things going. Is there any specific reason to write configuration in binary form since you have it anyway in your program?
See the code bellow:
void ICACHE_FLASH_ATTR CFG_Load()
{

	INFO("\r\nload ...\r\n");
/*	spi_flash_read((CFG_LOCATION + 3) * SPI_FLASH_SEC_SIZE,
				   (uint32 *)&saveFlag, sizeof(SAVE_FLAG));
	if (saveFlag.flag == 0) {
		spi_flash_read((CFG_LOCATION + 0) * SPI_FLASH_SEC_SIZE,
					   (uint32 *)&sysCfg, sizeof(SYSCFG));
	} else {
		spi_flash_read((CFG_LOCATION + 1) * SPI_FLASH_SEC_SIZE,
					   (uint32 *)&sysCfg, sizeof(SYSCFG));
	}
	if(sysCfg.cfg_holder != CFG_HOLDER){/**/
		os_memset(&sysCfg, 0x00, sizeof sysCfg);

		sysCfg.cfg_holder = CFG_HOLDER;

		os_sprintf(sysCfg.sta_ssid, "%s", STA_SSID);
		os_sprintf(sysCfg.sta_pwd, "%s", STA_PASS);
		sysCfg.sta_type = STA_TYPE;

		os_sprintf(sysCfg.device_id, MQTT_CLIENT_ID, system_get_chip_id());
		os_sprintf(sysCfg.mqtt_host, "%s", MQTT_HOST);
		sysCfg.mqtt_port = MQTT_PORT;
		os_sprintf(sysCfg.mqtt_user, "%s", MQTT_USER);
		os_sprintf(sysCfg.mqtt_pass, "%s", MQTT_PASS);

		sysCfg.security = DEFAULT_SECURITY;	/* default non ssl */

		sysCfg.mqtt_keepalive = MQTT_KEEPALIVE;

		INFO(" default configurations\r\n");

		CFG_Save();
//	}
	INFO("device_ID:%s\r\n",sysCfg.device_id);
	INFO("MQTTHOST:%s\r\n",sysCfg.mqtt_host);
}

Best regards,
K.

tuanpmt, 2015-01-31:
if you do like this, everytime boot, esp8266 will erase and rewrite default configurations, it would not be good for your memory with limit write count. if you want to change default configurations, please change CFG_HOLDER value

theater, 2015-01-31:
Hi,
Question is what if I take out cfg_save completely? Where else is this used
aside from initial boot?

tuanpmt, 2015-01-31:
it will load the default configuration every time boot and do not touch the flash memory

theater, 2015-01-31:
OK. I'll try to elaborate.

If we get rid of CFG_SAVE it will always put the values from user_config.h
into the memory and what we have in user_config.h will be loaded on every
reboot.
What is the reason to use CFG_LOAD in this case? Is it for having the
functionality of loading different configurations stored in different parts
of the memory?

Thanks.

tuanpmt, 2015-01-31:
it is used to load the configuration from Flash to RAM, if your application can change the configuration at runtime, whenever changes and call CFG_SAVE, then boot it will take this configuration

theater, 2015-01-31:
Understood. Thank you. Makes a lot of sense if you change configurations
runtime indeed.

closed 2015-01-31 by tuanpmt