tuanpmt/esp_mqtt #33: Problem with loading configurations

eliabieri, 2015-02-05:
Hello,
I compiled the firmware with my changed configurations in user_config.h and left the CFG_HOLDER unchanged(CFG_HOLDER	0x00FF55A4).
I then flashed the firmware and opened up the serial monitor.
This is the result:
what am I doing wrong?

Serial Results:

load ...
MQTT_InitConnection
MQTT_InitClient
WIFI_INIT

System started ...
mode : sta(18:fe:34:9d:5f:71)
add if0
STATION_IDLE
STATION_IDLE
STATION_IDLE
scandone
no DVES_HOME found, reconnect after 1s
STATION_IDLE
STATION_IDLE
reconnect
STATION_IDLE
STATION_IDLE
STATION_IDLE
STATION_IDLE
scandone
no DVES_HOME found, reconnect after 1s
STATION_IDLE
STATION_IDLE
reconnect

tuanpmt, 2015-02-05:
Try to change value of your CFG_HOLDER

eliabieri, 2015-02-05:
I changed it to 0x00FF55B, recompiled and reflashed the firmware but the result in the serial monitor is still the same. 
What can I do and what does DVES_HOME found mean?

PuceBaboon, 2015-02-05:
eliabieri,
   The DVES_HOME is Tuan's default access-point setting in the user_config.h which is shipped in the package.  You need to update that file with the specific details for your access-point (name of access point, userid and password) as well as specifying the MQTT server name (or IP) and port address.  I would recommend you leave the security settings alone until after you have the basic communications working.  Here's my /include/user_config.h file for reference:-

#ifndef _USER_CONFIG_H_
#define _USER_CONFIG_H_

#define CFG_HOLDER	0x00FF55A4
#define CFG_LOCATION	0x3C	/* Please don't change or if you know what you doing */

/*DEFAULT CONFIGURATIONS*/

#define MQTT_HOST	"het.pucebaboon.com"
#define MQTT_PORT	1883
#define MQTT_BUF_SIZE	1024
#define MQTT_KEEPALIVE	120	 /* In seconds. */

#define MQTT_CLIENT_ID	"Puce_%08X"
#define MQTT_USER	"Puce_USER"
#define MQTT_PASS	"Puce_PASS"

#define STA_SSID	"My_House"
#define STA_PASS	"0hN01t15nt"
#define STA_TYPE	AUTH_WPA2_PSK

#define MQTT_RECONNECT_TIMEOUT 	5	/* In Seconds. */

#define CLIENT_SSL_ENABLE
#define DEFAULT_SECURITY	0

#define QUEUE_BUFFER_SIZE 	512

eliabieri, 2015-02-06:
Thank you.
Now i see what DVES_HOME is.
I copied your user_config.h and filled in my STA_SSID and my STA_PASS.
I didnt change the CFG_HOLDER, recompiled and reflashed but in the serial monitor i can see that its still trying to connect to DVES_HOME. That means that my config file isnt loaded!

My second attempt was to change the CFG_HOLDER from the default(0x00FF55A4) to 0x00FF55A5
Then recompiled,reflashed but the problem stayed. 
In the serial monitor i can see that its still trying to connect to DVES_HOME

What am I doing wrong?

bcatalin, 2015-02-06:
Can you try to burn the blank.bin found in the bin directory of the SDK to
the address 0x3C000 ?

eliabieri, 2015-02-06:
Ok i found a workaround, it finally works now!
I changed the file config.c
Now:
if(sysCfg.cfg_holder = CFG_HOLDER)

Before:
if(sysCfg.cfg_holder != CFG_HOLDER)

The CFG_HOLDER is set to: x30C00

I think you see what I changed and the effect.
Thanks for you help!

closed 2015-02-06 by eliabieri

PuceBaboon, 2015-02-06:
Elia,

   Do a "make clean" before you do a recompile.