tuanpmt/esp_mqtt #34: functions not found

ranhaber, 2015-02-05:
Hi,
I have downloaded this great project.
In file mqtt.c there is a call for to function:
espconn_secure_disconnect()
espconn_secure_sent()
but there is no implementation.
If I'm wrong please direct me to it.
Thanks

PuceBaboon, 2015-02-05:
ranhaber,

   Take a look in your SDK directory for the file include/espconn.h.  You'll find the defines for the calls in there.  Make sure you're using SDK 0.9.4, as Tuan says in the readme file.

scargill, 2015-02-07:
Of interest I'm using SDK 0.9.5 without issue.

ranhaber, 2015-02-07:
are you using the make file attached here or make file from on of the examples?
When using the makefile attached I get this:
15:54:20 **** Build of configuration Release for project mqtt_ran ****
mingw32-make.exe -f C:/Users/HaberRan/workspace-mqtt/mqtt_ran/Makefile all 
CC driver/uart.c
CC mqtt/proto.c
CC mqtt/wifi.c
CC mqtt/utils.c
CC mqtt/config.c
CC mqtt/mqtt_msg.c
CC mqtt/queue.c
CC mqtt/ringbuf.c
CC mqtt/mqtt.c
CC user/user_main.c
AR build/app_app.a
LD build/app.out
FW firmware/0x00000.bin
process_begin: CreateProcess(NULL, c:/Espressif/xtensa-lx106-elf/bin/esptool -eo build/app.out -bo firmware/0x00000.bin -bs .text -bs .data -bs .rodata -bc -ec, ...) failed.
make (e=2): ‏‏The system can not find the file specified.

mingw32-make.exe: *** [firmware/0x00000.bin] Error 2
C:/Users/HaberRan/workspace-mqtt/mqtt_ran/Makefile:115: recipe for target 'firmware/0x00000.bin' failed

15:54:45 Build Finished (took 25s.401ms)

Can you help?

eadf, 2015-02-09:
Do you have c:/Espressif/xtensa-lx106-elf/bin/esptool  installed? 
It isn't installed by the https://github.com/pfalcon/esp-open-sdk scripts.

closed 2015-02-14 by tuanpmt