tuanpmt/esp_mqtt #36: Invalid protocol "MQTT" in CONNECT

bcatalin, 2015-02-05:
Hi, Tuan

 After updating with the last commit that resolved scargill issue with connectivity I've problems in connecting to the mosquitto broker. I guess that is the the same issue as in #35 since I have on my broker:

Feb  5 20:23:49 mosquitto[32712]: New connection from 192.168.1.14.
Feb  5 20:23:49 mosquitto[32712]: Invalid protocol "MQTT" in CONNECT from 192.168.1.14.

And from ESP
TCP: Connect to ip  192.168.1.26:1880
TCP: Reconnect to: 192.168.1.26:1880
MQTT: Connected to broker 192.168.1.26:1880
MQTT: Sending, type: 1, id: 0000
TCP: Sent
TCP: Disconnected callback
MQTT: Disconnected
Free memory

Can you help us with this one, please.

Regards,

Catalin

PuceBaboon, 2015-02-05:
Before anything else, just a big "Thank You!!" for all of the superb work you've done producing this package.  It really is a boon to the ESP8266 community.

On issue #36, I can verify that I'm seeing the same issue (invalid protocol "MQTT") using mosquitto broker under Linux using both a revision pulled from this site about six days ago and also with the latest pull.  The version of SDK used doesn't affect the error and I can also confirm that my build of mosquitto hasn't changed (mosquitto version 0.15 (build date 2013-01-31 17:51:33+0000)).

I cannot (so far) find any reference to protocol being set in the code (which I would have guessed would be set in the MQTT_client structure). Any ideas?

Again, very many thanks for such a useful package.

Best wishes from Japan,

                        -John-

tuanpmt, 2015-02-05:
Hello Jonh, 
Please see here: nodemcu/nodemcu-firmware#158

PuceBaboon, 2015-02-06:
Tuan,

   Many thanks for the speedy reply.  I'll try this fix (from Roger Light's
repository) now and put the details
   into the issue comments when I've verified it.

   Thanks again!

                                               -John-

PuceBaboon, 2015-02-06:
Tuan,

   Just a quick update... I ran into some incompatability problems when
trying to use Roger's
   repositories (I'm using a ElementaryOS, which is based on Ubuntu 12.05),
so instead I've
   just quickly installed a newer version of mosquitto (v1.3.5) under
FreeBSD on a Raspberry Pi
   server and I can confirm that the esp_mqtt client connects and publishes
without any
   problems.

   Have to go do some work now.  I'll try to find time to update the Linux
machine later.

   Thanks again!

                                                      -John-

closed 2015-02-06 by tuanpmt

asez73, 2015-02-06:
**Thank you for _all that work_!**
I actually preferred to upgrade Raspbian version of Mosquitto using the procedure described [here](http://mosquitto.org/2013/01/mosquitto-debian-repository/). 
This version of mosquitoo does comply with the standard version of MQTT on the broker's side.
Checking your mosquitto version of the MQTT protocol can be done with:
`sudo apt-cache search mosquitto`
You should see something like:
`mosquitto - MQTT version 3.1/3.1.1 compatible message broker`

This solved instanteanously the problem. 
Now the Mosquitto on Raspbian accepts the 3.1.1 protocol. I did not even reflashed the ESP8266.

The incompatibilties between broker's and client's versions of MQTT should be cared on the broker's side developpment but obviously, only from the standard 3.1.1,

bcatalin, 2015-02-06:
Hi, Is good to know that. I'll check to see what version of MQTT is
accepting my mosquitto on my PI.

Thanks,

Catalin

PuceBaboon, 2015-02-06:
Unfortunately, under Ubuntu and it's derivatives, the -bad- version, 0,15,
still shows up as
 "3.1/3.1.1 compatible".  :-(

bcatalin, 2015-02-06:
This is leaving me with no other option so I'll try to update the mosquitto
and check the MQTT code with version 3 and 4 to see if my update is ok for
my raspberry pi.

Thanks,

Catalin

PuceBaboon, 2015-02-06:
I've put up a How-To on configuring Ubuntu's Update Manager with Roger Light's repositories here:-

                    https://esp8266hints.wordpress.com/

and can confirm that mosquitto starts working again (as expected) with esp_mqtt.

tuanpmt, 2015-02-06:
Hello all,
I've add option to select version 3.1 or 3.11 in **user_config.h** 
```c
#define PROTOCOL_NAMEv31	/*MQTT version 3.1 compatible with Mosquitto v0.15*/
//PROTOCOL_NAMEv311			/*MQTT version 3.11 compatible with https://eclipse.org/paho/clients/testing/*/
```

bcatalin, 2015-02-06:
Thanks Tuan,

I will test it tonight and I will let you know.

Regards,

Catalin

PuceBaboon, 2015-02-07:
Excellent stuff!  Thanks Tuan.

scargill, 2015-02-07:
Is there an example of connecting to multiple MQTT servers. I note something about that in the readme file but I did not see an example. I just got my Raspberry Pi 2 and I'm itching to try Mosquitto on it.

Also.... that BLANK file you use at 0x3c000.  It erases the FIRST of your two blocks but am I right in saying it does NOT wipe the SECOND block... should you not be erasing the entire block up to 3ffff...

And finally, is there any reason you are not using 3d000 ?? If not I have a use for it...

commit 774f5a99 (2015-02-06): add protocol name selection(3.11 or 3.1), defaut is v3.1, resolve #36