tuanpmt/esp_mqtt #37: Hostname in localnetwork

aignaciors, 2015-02-06:
Hello,

I've been testing my mqtt network in a local network with fixed ip, but i have a dhcp server in a apache running in this network and i would like to know if is there a way to find the broker machine just with hostname in this LAN.

tks.

#define MQTT_HOST			"teste.local"

bcatalin, 2015-02-06:
I think that you can use the API function described in the SDK doc as:

Err_t espconn_gethostbyname(struct espconn *pespconn, const char
*hostname, ip_addr_t *addr, dns_found_callback found) to find the IP
address of your broker

Parameters:
struct espconn *espconn——corresponding connected control block
structure const char *hostname——domain name string pointer
ip_addr_t *addr——ip address
dns_found_callback found——callback

Return:

Err_t——ESPCONN_OK
ESPCONN_INPROGRESS
ESPCONN_ARG

Example as follows. Pls refer to source code of IoT_Demo:
ip_addr_t esp_server_ip;
LOCAL void ICACHE_FLASH_ATTR
user_esp_platform_dns_found(const char *name, ip_addr_t *ipaddr, void *arg)
{
struct espconn *pespconn = (struct espconn *)arg;
os_printf("user_esp_platform_dns_found %d.%d.%d.%d\n",
*((uint8 *)&ipaddr->addr), *((uint8 *)&ipaddr->addr + 1),
*((uint8 *)&ipaddr->addr + 2), *((uint8 *)&ipaddr->addr + 3));
}
Void dns_test(void)
{
espconn_gethostbyname(pespconn,”iot.espressif.cn”,&esp_server_ip,user_es
p_platform_dns_found);
}

Havving the IP address we can use it to pass it to the MQTT connect
function

Catalin

PuceBaboon, 2015-02-06:
Using the simple hostname ("teste"?) or the fully-qualified hostname (with the real domainname, such as "teste.example.com") will result in an automatic gethostbyname() lookup, which should work without any extra code.
Using the ".local" extension, as per your question, will probably not work unless you have something on your network which automatically responds to such requests with a real domain name.

aignaciors, 2015-02-06:
bcatalin :+1: 
I know this function, but the mqtt code from tuan already use this when you put the hostname from web, the problem is in the local network.

PuceBaboon:
Yes, it works for outside servers, but i`m using mDNS for my private network, and my broker does not has fixed ip. =/

PuceBaboon, 2015-02-06:
aignaciors,

I'm not aware of any mDNS capabilities within the SDK (but then, I know almost nothing about anything in the SDK :-))

You might want to check out the Avahi daemon, though.  It has the ability to "configure a unicast DNS server automatically from server data published on the LAN" and so should be able to automatically provide unicast answers to your multicast address request (I have a mixed environment here, with Macs co-existing in a mostly Un*x environment and Avahi works pretty well at handling the Mac mDNS traffic).  Of course, this all depends upon what you actually have in your environment and, more importantly, what you have administration rights on.

tuanpmt, 2015-02-14:
contiki have implemented mDNS support resolve local domain, you can look through. This project focus only mqtt protocol.

closed 2015-02-14 by tuanpmt