tuanpmt/esp_mqtt #39: Getting the received data

eliabieri, 2015-02-06:
Sorry guys for the noob question but maybe someone can help me.
I have a working esp_mqtt setup. I can publish data and subscribe to a topic.
My question is now:
How can I use the received data to turn a led on and off.
The received data gets processed in the void mqttDataCb() function.
The problem: I cant just do 
if(data=="on") {
      gpio_output_set(BIT0|BIT2, 0, BIT0|BIT2, 0);    //Set GPIO2 to HIGH / LED ON
}
What would be the right way?

tuanpmt, 2015-02-06:
Using strcmp function, lib string.h

eliabieri, 2015-02-06:
This is how I use it now: 
char key[] = "on";
	if(strcmp(key, data)){
      INFO("Turn SWITCH on!\r\n");
  }

Do i have to use data oder dataBuf here?
With the version above the strcmp is always true.

tuanpmt, 2015-02-06:
hello eliabieri,
strcmp return 0 if both strings is equal, and make sure it end with '\0'

eliabieri, 2015-02-06:
Ok this works now. 
I do it like this:

if(strstr(data, wordToCheck)) {
	INFO("matching string found'");
	}

closed 2015-02-06 by eliabieri