tuanpmt/esp_mqtt #40: Sample code for Arduino

Suxsem, 2015-02-10:
Hi! I'm trying to switch from nodemcu to this firmware...but i really can't understand how to use serial... can you please provide a basic example to send (and receive) mqtt messages from serial?

bcatalin, 2015-02-10:
Why do you need serial or Aduino ? The ESP will act as a publisher with a topic "/name/temp" for example and will send message you want. Every subscriber to this topic "/name/temp" will receive your message.

If you are the subscriber, you will receive every message for the subscribed topic. 

You can use mosquitto as a broker and mosquitto_pub ans mosquitto_sub to publish or subscribe to different topics.

Suxsem, 2015-02-10:
I know I know. But i want to use the esp8266 as a "bridge" to my arduino (because i prefer to code on Arduino and it has a lot more pins then esp8266-01 (my esp8266))
So i need serial :D

ystrem, 2015-02-10:
Hi, what exactly do you need ? MQTT to serial ?

Suxsem, 2015-02-10:
yes. now with nodemcu i wrote a small protocol to
1) connect to wifi from arduino
2) connect to mqtt broker from arduino
3) publish to topics
4) subscribe to topics
5) receive messages

something like
Serial.println(AT+wifi("ssid", "password");
Serial.println(AT+mqtt("mqttbroker", "password",...);
Serial.println(AT+subscribe("topic");
Serial.println(AT+publish("topic","message");
Serial.find("incoming message:");
.........

Suxsem, 2015-02-10:
something like the espduino https://github.com/tuanpmt/espduino but i want to leave the client on the esp8266 and use serial just to talk with the esp8266

Suxsem, 2015-02-10:
but i need just a basic code, an example of trasmit and receive from serial... i will implement all the functions

Suxsem, 2015-02-11:
any clue guys?

tuanpmt, 2015-02-14:
it should be another project, I think is okay to use SLIP protocol for transfer data between arduino and ESP8266

closed 2015-02-14 by tuanpmt

Suxsem, 2015-02-14:
but how?

tuanpmt, 2015-03-06:
Try this:
https://github.com/tuanpmt/espduino for arduino 
https://github.com/tuanpmt/esp_bridge for esp8266

Suxsem, 2015-03-06:
Wow great! THANKS!