tuanpmt/esp_mqtt #42: MQTT witch OTA

styszkiew, 2015-02-14:
This great but I can not get a library MQTT cope with OTA in the make file "ERROR build / app.out1 section` .irom0.text 'will not fit in the region `irom0_0_seg' 'if you can show your solution.

tuanpmt, 2015-02-14:
Need more info

styszkiew, 2015-02-14:
I wanted to add to the MQTT SUPPORT FOR OTA but I still do not know how to add the makefile productions user1.bin
my makefile looks like this:
BUILD_BASE	= build
FW_BASE		= firmware
FLAVOR = release
#FLAVOR = debug

# Base directory for the compiler
XTENSA_TOOLS_ROOT ?=  c:/Espressif/xtensa-lx106-elf/bin

# base directory of the ESP8266 SDK package, absolute
SDK_BASE	?= c:/Espressif/ESP8266_SDK_093

#Esptool.py path and port
PYTHON		?= C:/Python27/python.exe
SDK_TOOLS	?= c:/Espressif/utils
ESPTOOL		?= $(SDK_TOOLS)/esptool.exe
ESPPORT		?= COM2

JOIN_TOOL ?= $(SDK_BASE)/tools/gen_flashbin.py

WEB_DIR ?=C:\Espressif\examples\ota_test\www\htdocs\v1\device\rom

# name for the target project
TARGET		= app

# which modules (subdirectories) of the project to include in compiling
MODULES		= driver mqtt user user/chip user/io
EXTRA_INCDIR    = include $(SDK_BASE)/../include

# libraries used in this project, mainly provided by the SDK
LIBS		= c gcc hal phy pp net80211 lwip wpa upgrade main ssl

# compiler flags using during compilation of source files
CFLAGS		= -Os -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals  -D__ets__ -DICACHE_FLASH

# linker flags used to generate the main object file
LDFLAGS		= -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static

ifeq ($(FLAVOR),debug)
    CFLAGS += -g -O2
    LDFLAGS += -g -O2
endif

ifeq ($(FLAVOR),release)
    CFLAGS += -g -O0
    LDFLAGS += -g -O0
endif

# linker script used for the above linkier step
LD_SCRIPT	= eagle.app.v6.ld
LD_SCRIPT1	= eagle.app.v6.app1.ld
LD_SCRIPT2	= eagle.app.v6.app2.ld

# various paths from the SDK used in this project
SDK_LIBDIR	= lib
SDK_LDDIR	= ld
SDK_INCDIR	= include include/json

# we create two different files for uploading into the flash
# these are the names and options to generate them
FW_FILE_1	= 0x00000
FW_FILE_1_ARGS	= -bo $@ -bs .text -bs .data -bs .rodata -bc -ec
FW_FILE_2	= 0x40000
FW_FILE_2_ARGS	= -es .irom0.text $@ -ec

FW_FILE_3	= user1
FW_FILE_4	= user2

#Intermediate files for User1.bin and User2.bin
[... 168 lines cut ...]

$(foreach bdir,$(BUILD_DIR),$(eval $(call compile-objects,$(bdir))))

Crashes when compiling error:
make all 
CC driver/uart.c
CC mqtt/config.c
CC mqtt/mqtt.c
CC mqtt/mqtt_msg.c
CC mqtt/proto.c
CC mqtt/queue.c
CC mqtt/ringbuf.c
CC mqtt/utils.c
CC mqtt/wifi.c
CC user/user_main.c
CC user/chip/chipother.c
CC user/chip/dht.c
CC user/chip/ds18b20.c
CC user/chip/work_led.c
CC user/io/io.c
AR build/app_app.a
LD build/app.out
c:/espressif/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: build/app.out1 section `.irom0.text' will not fit in region `irom0_0_seg'
collect2.exe: error: ld returned 1 exit status
make: *** [build/app.out] Error 1

tuanpmt, 2015-02-14:
When using OTA and booloader then the memory partition only one half. I think your program is too much for that. Please see #31 if your functions not have ICACHE_FLASH_ATTR attribute

styszkiew, 2015-02-14:
How to compile with the library esp_mqtt to user1.bin it also shows the ".irom0.text' will not fit in regionirom0_0_seg".
What do I have to change in order to compile the esp_mqtt user1.bin?

tuanpmt, 2015-05-15:
ref: #57

closed 2015-05-15 by tuanpmt