tuanpmt/esp_mqtt #43: Very, very minor update to README.md

PuceBaboon, 2015-02-16:
Apologies for the issue (still haven't worked out how to use git properly).

Thanks for the updates to the Makefile and to the README.md.  I'd like to suggest a very small update to the README.md which might be more realistic for Linux/Mac users:-

----------------------------------------------------------------------------------
for Mac or Linux:

```bash
git clone https://github.com/tuanpmt/esp_mqtt
cd esp_mqtt
#clean
make clean
#make
make SDK_BASE="/opt/Espressif/ESP8266_SDK" FLAVOR="release" all
#flash
make ESPPORT="/dev/ttyUSB0" flash
```
----------------------------------------------------------------------------------

closed 2015-02-16 by tuanpmt

PuceBaboon, 2015-02-16:
Tuan,

   What's the best way to send you very minor updates like this?  What's
easiest for you?

                                     -John-

tuanpmt, 2015-02-16:
First, you need to fork this project. Next, you can pull update to your repository.
When you fixes some issues, you commit hand push to your repo, then, create a pull request to this repo.

PuceBaboon, 2015-02-16:
Okay, got it!   Thank you.

       -John-

commit 10d13ed5 (2015-02-16): Update README.md, resolve #43