tuanpmt/esp_mqtt #44: wifi.* and config.* under user module

eadf, 2015-02-16:
As you know I'm using mqtt as a library component. 
I do this by simply adding **path_to_mqtt/mqtt** to the MODULES variable. What i can't do is to add **path_to_mqtt/user** to MODULES since then there will be two user_init() functions in the bin.
This used to work, but since you moved wifi.h, wifi.c, config.h and config.c to **path_to_mqtt/user** my builds fail.

I suggest you move those files back to  **path_to_mqtt/mqtt** or create another module folder.

tuanpmt, 2015-02-16:
I know that eadf, but in other applications, they don't need wifi and config , i think we can move it to another folder like **path_to_mqtt/modules**. What do you think?

eadf, 2015-02-16:
that would be super, thanks

closed 2015-02-16 by tuanpmt