tuanpmt/esp_mqtt #45: Full retransmit support for QoS1 and QoS2

Suxsem, 2015-02-16:
Hi. I found in the readme:
Limited:
Not fully supported retransmit for QoS1 and QoS2

It's this still true?
Can you confirm that this problem also affects lua port of esp_mqtt (nodemcu)?

Because sometimes I get the published callback on the esp8266 but the broker doesn't receive the message... This occurs when i send a lot of messages in short time.

tuanpmt, 2015-02-16:
you can see here: https://code.google.com/p/mqtt4erl/wiki/QualityOfServiceUseCases
We need outbox for publish message with Qos=1 and 2, We will only remove it when you receive a PUB ACK, if not, we need retransmit message.
Currently, the library do not support that behavior.

If you send too many messages in a short time, you should make sure it does not exceed capacity of the Queue. it is FIFO, will remove the oldest message to be room for the latest message.

Suxsem, 2015-02-16:
ok, thanks :)

TheGuv, 2015-03-17:
Hello.
Are they any plans to fully support QoS 1 and 2? I understand an outbox would need to be crafted in, but I would really appreciate full support due to the massive amount of radio interference at my location which often causes missing messages.

tuanpmt, 2015-03-17:
I have no plans to do so

TheGuv, 2015-03-23:
Oh, that's a shame. Doesn't the lack of support mean that delivery is not guaranteed, or does it mean there may be more than one delivery of a message? Please would you confirm what the current support provides.

tuanpmt, 2015-03-23:
Now it only guaranteed feedback PUBREL, PUBCOM from server, meaning guarantee direction from Server

TheGuv, 2015-04-05:
A pity indeed as it makes, for example, motion sensors unreliable. They only have one job to do (report motion) and it seems this can't be guaranteed which kind of defeats their purpose. Thanks anyway and perhaps one day this might be revisited?