tuanpmt/esp_mqtt #46: Is SSL possible?

jdunmire, 2015-02-22:
I have not been able to get an SSL connection to work between the esp8266 and mosquitto running on Ubuntu 14.04. Is SSL really working? If so, what steps are needed?

I am using the HEAD of the master branch and compiling with esp-open-sdk+esp_iot_sdk_v0.9.5.
I set DEFAULT_SECURITY to 1 in user_config.h.
I also instrumented the code to confirm that that the espconn_secure* functions are called.

I have tried using 512, 1000, 1024, and 2048 bit SSL certs on the server.

tuanpmt, 2015-03-23:
Try to build with the sdk 0.9.3 here : http://www.esp8266.com/viewtopic.php?p=5979#p5979

jdunmire, 2015-04-01:
SDK 0.9.3 works, thank you.
I also tried the recently released 1.0.0 SDK and it still does not work.

EspressifSystems, 2015-04-03:
will look into this.

abl, 2015-04-15:
I had better luck with not just SDK 0.9.3 but also the patched libssl from Espressif that added a larger buffer:

http://bbs.espressif.com/viewtopic.php?f=7&t=62

There's also some action on the latest (1.0.1b2) SDK related to SSL support:

http://bbs.espressif.com/viewtopic.php?f=7&t=365&sid=027421e52f1e9551e74182e831e35d94

Tylerflick, 2015-05-13:
I've pulled the libssl patch in from Espressif's website but it hasn't seemed to fix this TLS bug for SDK v1.0.1. Is anyone here trying to build this with esp-open-sdk?

tuanpmt, 2015-05-13:
yes, i've tried and it work ok, but need more than 17Kb RAM

Tylerflick, 2015-05-13:
@tuanpmt Was that with the example NodeJs Server?
I haven't had any luck connecting to my local Mosquitto server running a 102

tuanpmt, 2015-05-13:
with google https and pushbullet https, just handshake ok and disconnect

Tylerflick, 2015-05-14:
I'm still seeing `error:140760FC:SSL routines:SSL23_GET_CLIENT_HELLO:unknown protocol.` in my server logs when the chip tries to connect.

In my user_config.h I have `#define CLIENT_SSL_ENABLE 1` & `#define DEFAULT_SECURITY 0`.
Is there anything else I need set?

tuanpmt, 2015-05-15:
change ```#define DEFAULT_SECURITY 1```

Tylerflick, 2015-05-15:
Yeah I caught that mistake, but it didn't fix my issue.

Digging around in the SDK I've noticed a few of the example programs reference a cert.h generated from the servers TLS cert. Is this something new that we need?

tuanpmt, 2015-05-15:
No, only https server need

Tylerflick, 2015-05-15:
Okay, I'll wait for a patch from @EspressifSystems. Thanks for your help with this.

Tylerflick, 2015-05-17:
@tuanpmt I've been able to get the clients communicating with a mosquitto broker over TLS. Thanks!

On a side note, using the 1.0.1 SDK with some 1-wire example code provided by @scargill compiles but will not run. I suspect the device is running out of memory.

scargill, 2015-05-17:
Not sure which device you're using here- but if you're short of FLASH, remember most of the ESP-12s have much larger FLASH than the ESP-01 and ESP-07.

Tylerflick, 2015-05-17:
I just read your blog post on that. I rebuilt the program using v0.9.3 of the SDK and everything seems to be functioning fine.

scargill, 2015-05-17:
Interesting – I have had issues with ESP-01 boards and ESP-07 of code not running – which will run on the ESP-12… but no problems using SDK versions before 1.0

 

I solved mine by reversing the order that the FLASH is blown – ie the 0x40000 first and 0 last.  Someone commented something about the way that sector erasing happens and some overlap but I don’t know enough about that to comment further right now – would be nice to hear from people who have had this issue.

closed 2015-05-30 by tuanpmt