tuanpmt/esp_mqtt #50: Compile failed on Ubuntu LTS 14.04

nmaas87, 2015-03-08:
I was using esp-open-sdk ( https://github.com/pfalcon/esp-open-sdk ) and was getting this error:

make SDK_BASE="/home/pi/esp/esp-open-sdk" FLAVOR="release" all
CC driver/uart.c
CC mqtt/mqtt.c
CC mqtt/ringbuf.c
CC mqtt/mqtt_msg.c
CC mqtt/queue.c
CC mqtt/proto.c
CC mqtt/utils.c
CC user/user_main.c
CC modules/config.c
CC modules/wifi.c
AR build/app_app.a
LD build/app.out
/home/pi/esp/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot open linker script file /home/pi/esp/esp-open-sdk/ld/eagle.app.v6.ld: No such file or directory
collect2: error: ld returned 1 exit status
Makefile:184: recipe for target 'build/app.out' failed
make: *** [build/app.out] Error 1

aignaciors, 2015-03-08:
Verify you PATH variable and your SDK folder to confirm that you have the files like eagle.app.v6.ld....

nmaas87, 2015-03-08:
Sorry, that was a really silly question.
The path I gave was not really the SDK_BASE, but the whole toolchain folder. Got that wrong.
The correct call for the esp-open-sdk is following:

make SDK_BASE="/home/pi/esp/esp-open-sdk/sdk" FLAVOR="release" all

And with that, compilation works :)!

Thanks a lot!

closed 2015-03-08 by nmaas87

Caerbannog, 2016-04-10:
This should now be fixed.