tuanpmt/esp_mqtt #51: error in reading config - ESP8266-01

terenceang, 2015-03-12:
My ESP8266-01 won't connect to my AP.

was getting "no 1 found, reconnect after 1s"

Apparently, the 1 is loaded into SSID on CFG_Load(), getting defaults loaded don't seems to work too.
A temp fix is to force the SSID from the header.

		os_sprintf(sysCfg.sta_ssid, "%s", STA_SSID);

OliverHi, 2015-03-13:
I can confirm that problem.

tuanpmt, 2015-03-14:
when you change default configuration, please change value of CFG_HOLDER, or write blanker value to address 0x3C000

georgkreimer, 2015-05-05:
What would you set CFG_HOLDER to? Any best practice? I'm using ESP8266 12.

tuanpmt, 2015-05-05:
any value, it used for check first time write default configuration, if you change it, all values reset to default after esp reset

georgkreimer, 2015-05-05:
Ok, cool! Just making sure it doesn't write to wrong places ;) Thanks!

closed 2015-05-15 by tuanpmt