tuanpmt/esp_mqtt #53: After a few messages wifi reconnects...

dmarkey, 2015-04-12:
Like so:

Receive topic: /topic, data: test
rm match
TCP: Reconnect to mosquitto:8000
reconnect
scandone
add 0
aid 2
cnt

connected with Markey, channel 10

Seems to happen every 20-30 messages received.

harryd100, 2015-04-14:
If I use a Linksys with DD-WRT I have the same issue. Using an old Netgear WiFi router keeps me connected for several days now.

tuanpmt, 2015-04-14:
try to build with sdk 1.0.1, i don't known what happen, but with logs:
```
reconnect
scandone
```
Cause from SDK

harryd100, 2015-04-14:
Already done building with 1.0.1 SDK, same problem. Going to upgrade my router soon and test again as the Netgear router seems to work fine.

dmarkey, 2015-04-14:
could it be a g vs n problem?

harryd100, 2015-04-14:
@dmarkey You could be right. If I put the Linksys router in N or G mode only, it stays connected. As soon as I put it in NG mixed mode, occasionally a reconnect occurs. Tried 2 ESP-01 modules, both seem to behave the same. I'm not a WiFi expert, so don't know why this happens.

tinamore, 2015-05-19:
Same problem, ESP publish a few minutes, after can not send. This a problem, becase is a basic IOT.

Werquinx, 2015-05-19:
Same problem. Made a logging unit. Worked fine while testing but when put in real life situation it doesn't work as the unit seems to disconnect for no obvious reason and stops sending MQTT messages after a while ... If there is no way to detect this and take action on it (reconnect or resubscribe) then we cannot build reliable IOT devices :-/

tinamore, 2015-05-19:
I think is has found the solution.

Currently, ESP does not good in mode "PHY_MODE_11N". If ESP8266 run on network g/n, it will crash and reboot regularly.

Project tuanpmt esp_mqtt has not config PHY_MODE for ESP8266.

I've added: wifi_set_phy_mode (PHY_MODE_11B) on function user_init in the esp_mqtt example in the SDK Espressif, now it runs very good. No restart, Send messenger is very good.

Werquinx, 2015-05-19:
Where did you call it exactly ?

I modified as follows : 
	wifi_set_phy_mode (PHY_MODE_11B);
	WIFI_Connect(sysCfg.sta_ssid, sysCfg.sta_pwd, wifiConnectCb);

But this doesn't have any effect for me, still stops responding after 10 minutes ....

Thanks.

tinamore, 2015-05-20:
Hi
I just test 5 minutes, previously only run 1 minute then stop.
I will test more time and reply.

PuceBaboon, 2015-05-20:
Xavier,

   Some people have reported success by changing their access point (the
one which the ESP8266 attaches to, not the ESP8266 itself) from
auto-selection of radio mode to manual selection of just B, G or N.  This
didn't work for me, though.

   My AP (an older model FON running DD-WRT) is now configured to reboot
every night, otherwise it loses the ARP cache entries for the ESP8266 and
for a wireless HP printer.  All of the other machines on the wireless side
of the network seem to be fine though, so they seem to be able to perform
some sort of recovery that the HP and the ESP do not.

   You might want to check the ARP cache on your MQTT server (use "arp -a")
and then, if it's possible with your AP, check the wireless associations to
see whether the ESP is still associated, too.  This should at least give
you a clue as to whether its the ESP or the AP which is the problem.

                                                                 -John-

Werquinx, 2015-05-20:
I already tried on 2 different APs and same issue ...

For me an iot device should be plug and play. I have many devices connected to 4 different APs in the house and all connect reliably ... As such I can only expect in an issue in the ESP8266 stack ?

Xavier

Werquinx, 2015-05-20:
By the way. the module seems to run OK as I have a led which indicates sensor status based on GPIO input and it continues to operate correctly. It is just the MQTT messages that don't get sent ...

harryd100, 2015-05-20:
@Werquinx, strange behaviour. I had some problems too with ESP's disconnecting from WiFi when the AP was in mixed or N/G mode, but never from MQTT (Mosquitto). After changing the ESP code to use PHY_MODE_11B, I could run my AP's in mixed mode and since that time it is stable for days now using TuanPM's MQTT code.

Werquinx, 2015-05-20:
You can see my main file here :
https://github.com/Werquinx/esp_mqtt_eauchaude/blob/master/user/user_main.c

Am I doing something wrong ? Is the call to PHY_MODE in the right place ?

Thanx

harryd100, 2015-05-20:
I do it this way, first init WiFi, the open connection to broker

wifi_set_phy_mode (PHY_MODE_11B);
WIFI_Connect(sysCfg.sta_ssid, sysCfg.sta_pwd, wifiConnectCb);

and then do the MQTT stuff

MQTT_InitConnection	(&mqttClient, sysCfg.mqtt_host, sysCfg.mqtt_port, sysCfg.security);
MQTT_InitClient		(&mqttClient, sysCfg.device_id, sysCfg.mqtt_user, sysCfg.mqtt_pass, sysCfg.mqtt_keepalive, 1);
MQTT_InitLWT		(&mqttClient, "/lwt", sysCfg.device_id, 0, 0);
MQTT_OnConnected	(&mqttClient, mqttConnectedCb);
MQTT_OnDisconnected	(&mqttClient, mqttDisconnectedCb);
MQTT_OnPublished	(&mqttClient, mqttPublishedCb);
MQTT_OnData		(&mqttClient, mqttDataCb);

Werquinx, 2015-05-23:
I did some more tests and based on indications found around the web it was mentioned that the ESP is not stable in station only mode in some wifi environments.

So I changed wifi.c and changed to station + AP mode, cf : 
	wifi_set_opmode(0x03);

Doing this, I managed to have my module stay connected for 6 hours in stead of the typical 15 minutes ...

But it still disconnects at some time :-/     (and creates the useless AP network).

It seems to me that ESP8266 is not ready for prime time yet ?

harryd100, 2015-05-27:
Upgraded to SDK 1.1.0 yesterday and all ESP-01 still connected to the AP (13 hours now)

dmarkey, 2015-05-27:
Nice!

tuanpmt, 2015-05-30:
how it's going @harryd100 ?

harryd100, 2015-05-30:
@tuanpmt, once connected they stay connected for days now. I have a second AP (for coverage) with the same SSID on another channel. If I reboot the router where the ESP's are connected to, sometimes they will not roam to the other AP or when they do roam MQTT is not working. I think the code is still running because the WDT is not fired. The ESP not roaming is a SDK thing I believe. I'm thinking of detecting the AP change (channel change) and do a node_restart. When the AP is connected the MQTT code works flawless, i'm very happy with that. Thanks for this nice feature piece of MQTT code.

tuanpmt, 2016-09-09:
@harryd100 It seems there is a limitation of the SDK, a long time ago, but I still want to ask how you deal with it?

harryd100, 2016-09-11:
@tuanpmt, I'm running various ESP-01, 07 and 12(E) devices with SDK 1.5.2 and your MQTT code for a very long time now and must say it's very stable at the moment. Even when I reboot one of the AP's, the ESP's connected to that one roam to the other if they are in range. I recently converted 3 (Sonoff) switches, also  using ESP's, to SDK 2.0.0 with the latest MQTT code and monitoring that behaviour. But at first glance it looks very promising. In the past if there where problems, you would notice in hours. I also removed the wifi_set_phy_mode (PHY_MODE_11B) which I needed in the passed to get a stable WiFi connection.

closed 2016-09-12 by tuanpmt