tuanpmt/esp_mqtt #54: Mosquitto mqtt broker on Rasberry Pi 2 connection issue

chumxeed, 2015-04-26:
I just finish installing esp8266 tool-chain and try to get familiar with your esp_mqtt. so this is my very first program I compile and flash to the esp-07 module. the flashing was a success  but i having an issue connecting to my mosquitto mqtt broker running on rasberry pi 2. the following message constantly display in serial terminal. could you help take a look on my issue ?. BTW. i just install mosquitto on my ubantu with the same procedure as the one installed on rasberry pi. there is no issue at all to connect so I'm not sure what causing this.

TCP: Connect to ip  172.16.1.22:1880
TCP: Reconnect to: 172.16.1.22:1880
MQTT: Connected to broker 172.16.1.22:1880
MQTT: Sending, type: 1, id: 0000
TCP: Sent
TCP: data received 4 bytes
MQTT: Connected to 172.16.1.22:1880
MQTT: Connected
MQTT: queue subscribe, topic"/mqtt/topic/0", id: 21
MQTT: queue subscribe, topic"/mqtt/topic/1", id: 22
MQTT: queue subscribe, topic"/mqtt/topic/2", id: 23
MQTT: queuing publish, length: 23, queue size(646/2048)
MQTT: queuing publish, length: 25, queue size(671/2048)
MQTT: queuing publish, length: 25, queue size(698/2048)
TCP: Disconnected callback
MQTT: Disconnected
Free memory

chumxeed, 2015-04-26:
I did enable debug log in rpi and found that mosquitto was refusing connection from esp_mqtt. it sening connack with return code 5. then i guess it close the socket immediately after that. but i can connect to the rpi with mqtt spy with the same user and password. please advice how i can resolve this issue.
thanks in advance.

chumxeed, 2015-04-29:
I have identify some differences between my mosquitto installation.
the one that install in my virtual machine under ubantu was mosquitto 1.4 
on my rpi 2 installation install using standard apt-get install was mosquitto 1.35
so I try again to identify whether the issue still remain there or not by install a pi snappy ubuntu image and perform installation of mosquitto using snappy but I can not configure my mosquitto to be the same as my ubantu installation. so I take another route by downloading necessary liblary and deb package for mosquitto from whizzy debien repo then install it with dpkg -i then i can configure it but the problem still the same. i guress i need mosquitto 1.4 for testing but there is no rasberry pi 2 binary exis in any repo yet so I may need to go deeper again to do cross compiling mosquitto on my existing unbuntu. BTW. if you have any work around or solution on esp8266 end please update or post the solution here I'd very greatfull.

tuanpmt, 2015-04-29:
maybe you need to change 
```c
#define PROTOCOL_NAMEv31	/*MQTT version 3.1 compatible with Mosquitto v0.15*/
```
from user_config.h

chumxeed, 2015-04-29:
I got it working now Thanks, for the PROTOCOL_NAME setting I did not change anything yet, just clone your git repo them make and flash to the module so I'm not sure what is the actual problem.
here is my solution.
I Download Rasbian Debian Wheezy fresh from  https://www.raspberrypi.org/downloads/ then create a boot sd for my raspberry pi 2 then.
1. update the pi to latest software
    $ sudo su
    # apt-get update && apt-get upgrade
2. Install Development Environment on Pi 2
    # apt-get install build-essential
3. Download mosquitto 1.4 source tar file from http://www.eclipse.org/downloads/download.php?file=/mosquitto/source/mosquitto-1.4.1.tar.gz (iput it on  pi's home directory)
4. extract tar -xvzf  <tarfile> it should create a directory mosquitto-1.4.1
   # cd mosquitto-1.4.1 
   # make
5. I got error during make cause by missing uuid.h  if you got error like me you need to install uuid-dev
   # apt-get install uuid-dev
6. # make install
after that config mosquitto as it should be also if you have a script to start mosquitto pit it in /etc/init.d 
That conclude my problem

closed 2015-04-29 by chumxeed