tuanpmt/esp_mqtt #60: Error pointer targets in assignment differ in signedness

tinamore, 2015-05-28:
Hi, I run esp_mqtt and show error in elipse:
----
mqtt/utils.c: In function 'UTILS_StrToIP':
mqtt/utils.c:100:12: error: pointer targets in assignment differ in signedness [-Werror=pointer-sign]
      start = str;
            ^
cc1.exe: all warnings being treated as errors
C:/Espressif/examples/esphttpd/Makefile:442: recipe for target 'build/mqtt/utils.o' failed
mingw32-make.exe: *** [build/mqtt/utils.o] Error 1
---
Help me fix this error, thanks verymuch

tuanpmt, 2015-05-28:
may you modify `const char * start;` to `const int8_t * start;` will work

tinamore, 2015-05-28:
yeah, working now, thanks.
I really admire you.
I'm in Ho Chi Minh City, I can invite you coffee :D?

tuanpmt, 2015-05-30:
Wellcome @tinamore , i'm in District 9, when you ready, please email  to me.

closed 2015-05-30 by tuanpmt