tuanpmt/esp_mqtt #62: Compiling with ESP8266 SDK (esp_iot_sdk_v1.1.0_15_05_26) gives error

viktak, 2015-06-04:
I have been using this module since version 0.9x. I just updated my dev box to v1.1.0 and my code doesn't compile any more. This may not be the best forum to post this issue, but I'm new to github, so please excuse me for that.
This is the output of the compilation:

```
CC driver/uart.c
CC mqtt/mqtt.c
CC mqtt/ringbuf.c
CC mqtt/mqtt_msg.c
CC mqtt/queue.c
CC mqtt/utils.c
CC mqtt/proto.c
CC user/user_main.c
CC modules/config.c
CC modules/pwm.c
CC modules/wifi.c
AR build/app_app.a
LD build/app.out
/opt/Espressif/esp_iot_sdk_v1.1.0/lib/libmain.a(app_main.o): In function `user_uart_wait_tx_fifo_empty':
(.irom0.text+0x34c): undefined reference to `user_rf_pre_init'
/opt/Espressif/esp_iot_sdk_v1.1.0/lib/libmain.a(app_main.o): In function `user_uart_wait_tx_fifo_empty':
(.irom0.text+0x474): undefined reference to `user_rf_pre_init'
collect2: error: ld returned 1 exit status
Makefile:184: recipe for target 'build/app.out' failed
make: *** [build/app.out] Error 1
```

bcatalin, 2015-06-04:
add

void user_rf_pre_init(){}

to your user_main.c file

Catalin

viktak, 2015-06-04:
When I do this it compiles fine, but when I upload the firmware to the ESP it doesn't run. Looking at the UART of it it only does the usual garbage (I know it's not garbage, but a different speed) at power on, but then nothing else.

bcatalin, 2015-06-04:
I have the same problem. I'll let you know about this, or I'll post it to
my blog myesp8266.blogspot.com

bcatalin, 2015-06-04:
Is your SDK an upgrade to an existing one, or is a fresh install ?

viktak, 2015-06-04:
It is a fresh install of the SDK, but the rest of the toolchain is the same. What I did was I installed the new one next to the previous ones and I just edited the SDK's path in the command line I used for compiling the project.
So as far as I understand the process this counts as a fresh install of the SDK, not an upgrade.

bcatalin, 2015-06-04:
I've tried with fresh install and as an upgrade to a previous SDK. Both
ways had the same result, garbage on serial.

I am suspecting that the files are not written to the correct addresses in
the flash.

Peter Scargill http://tech.scargill.net/ had a similar problem and in his
case the guilty file was the eagle.app.v6.ld ( located in the ld
directory). He used an old version of that file.

I've compared his file with mine and are the same.

Please let me know if you find something on this issue. I'm blocked right
now with it.

Regards,

Catalin

bcatalin, 2015-06-04:
Hi Tuan,

May be you can help me since I am blocked, and I don't see any light.
I've recompiled the esp-open-sdk, put your code and compiled. Everything was fine, but the code
is not running. I've receive on serial only rubbish characters.

Output:
rll|l|lb|rblbnnlnnbplrlrlrnbnbp

If you will have some time to help us with this issue will be nice. From my point everyone that
is using esp-open-sk and your mqtt will fail in running it.

Regards,

Catalin

tuanpmt, 2015-06-05:
i've tried to compile esp_mqtt with sdk 1.1.0 and have same problem with you, their sdk development is not backward compatible. Quick solution: compile with sdk 1.0.1 and using some new lib of sdk 1.1.0: libssl, libsmart ...
I will try to make esp_mqtt can use with the latest sdk

bcatalin, 2015-06-05:
Thanks, I will try it.

harryd100, 2015-06-05:
Hmm, this is strange, i'm running several devices witch TuanPMT's MQTT code and have no problems at all. However, I did install the latest Espressif dev-kit and it is working fine

bcatalin, 2015-06-05:
Please tell us all the steps you did to upgrade from 101 to 110.

harryd100, 2015-06-05:
Just install the new SDK in a separate directory and change your makefile accordingly. I use Eclipse under Linux so you have to tell the compiler where to find the include files for Espressif and the new SDK.

bcatalin, 2015-06-08:
Changing the flashing file order solved the problem. Anyway the problem must be investigated to see way the flash get corrupted.

So first flash the data at 0x40000 and then the frash.bin at 0x00000.

I've checked the version reported by ESP and indeed is the latest one.

SDK version: 1.1.1 
sntp: 1433808934, Tue Jun 09 00:15:34 2015

My command is sudo /home/esp8266/COD/esp-open-sdk/esptool/esptool.py --port /dev/ttyUSB0 write_flash 0x40000 /home/esp8266/PROJECT/irritool/firmware/0x40000.bin 0x00000 /home/esp8266/PROJECT/irritool/firmware/0x00000.bin

Catalin

viktak, 2015-06-08:
Yes, I can confirm this workaround works. I used NodeMCU Firmware Programmer 64-bit under Windows 8.1 and changing the file order fixed the issue.

However, I don't understand why this works. I was under the impression that the order did not matter, only the content. After all, we are simply saving data...

Thanks for sticking with me on this issue and providing me at least a workaround!

viktak

bcatalin, 2015-06-08:
Even the addresses are far away, the problem can be on erasing the the
flash segment.

johnnyquad, 2015-06-09:
Hi there,
I have swapped the flashing order and am using 1.1.0, have correctly set  user_config.h with
#define STA_SSID "myLocalSSID"
#define STA_PASS "myPassword"
#define STA_TYPE AUTH_WPA2_PSK

all I get is 
STATION_IDLE
scandone
no _00988692 found, reconnect after 1s
reconnect
f 0, STATION_IDLE
STATION_IDLE
STATION_IDLE
STATION_IDLE
scandone
no _00988692 found, reconnect after 1s
STATION_IDLE
reconnect
f -240, STATION_IDLE
STATION_IDLE
STATION_IDLE
STATION_IDLE
scandone
no _00988692 found, reconnect after 1s
reconnect
f 0, STATION_NO_AP_FOUND
scandone
f -240, STATION_IDLE

repeating forever

Maybe this is not a SDK issue bit I don't know what to look at next , what does "no _00988692 found" mean ?(sorry if this is in the wrong place)

john

db1mso, 2015-06-10:
Try to flash each bin separately.
In my case, there is a target "flashonefile" and "flashinit" in makefile.
This workaround worked for me.

olli

tuanpmt, 2015-06-11:
@johnnyquad, change value of CFG_HOLDER may help you

kanflo, 2015-06-30:
@tuanpmt, I guess this issue can be closed. Regarding the flashing order, that will be taken care of in esptool.py (see https://github.com/themadinventor/esptool/pull/51)

viktak, 2015-06-30:
Still, a note must be added somewhere to the kit as not everyone uses esptool.py. I use NodeMCU Firmware Programmer under Windows.

kanflo, 2015-07-01:
@viktak, you're right. I added an issue about this to the NodeMCU FP.

koltegirish, 2015-10-29:
Hello Guys,
When I try to compile code using Mesh lib it shows following error .Any guest how to resolve it?What will be the issue?
AR build/test_app.a
LD build/test.out
c:/Espressif/ESP8266_SDK/lib\libmain.a(app_main.o): In function `user_uart_wait_tx_fifo_empty':
(.irom0.text+0x598): undefined reference to `user_init'
c:/Espressif/ESP8266_SDK/lib\libmain.a(app_main.o): In function `user_rf_pre_init':
(.irom0.text+0x804): undefined reference to `user_init'
collect2.exe: error: ld returned 1 exit status
mingw32-make.exe: *** [build/test.out] Error 1
C:/Users/ABC/worksorg/mesh/Makefile:187: recipe for target 'build/test.out' failed

11:33:15 Build Finished (took 1s.242ms)

Deadolus, 2016-01-15:
I know this is an old thread, but posting to update this with new info and my own experiences with the ESP8266 - maybe some poor soul will profit in the future. 

**This project compiles fine for me with SDK 1.5.0 after applying this patch https://github.com/tuanpmt/esp_mqtt/pull/91**

It also gives me non-garbage output on my serial console. So the problem is problem with your setup, if something does not work. 
@koltegirish @viktak The undefined references come from a linker problem (as you might have guessed). So check your linker file (the .ld file in the SDK), check that you have all modules (this was the problem for me when updating to the latest SDK) and check that your linking the right .o files

@johnnyquad This means the ESP did not find a Access Point with this SSID. Check that you SSID was not overwritten by something while compiling (maybe search for the text "_00988692") and that your "mylocalSSID" does not contain any unprintable or not allowed characters

tuanpmt, 2016-09-09:
Updated to SDK 2.0, please check develop branch, it will be merge soon

closed 2016-09-09 by tuanpmt