tuanpmt/esp_mqtt #63: Issue connecting to broker on raspberry pi

rhouri, 2015-06-18:
I'm getting the following resonse on the serial comms. I upgraded mosquitto on the pi to 141, and did all the fixes I could find to no avail...
Any one knows what's going on ?
Thanks 

load ...
MQTT_InitConnection
MQTT_InitClient
WIFI_INIT

System started ...
mode : sta(18:fe:34:98:6a:e6)
add if0
f 0, STATION_IDLE
STATION_IDLE
STATION_IDLE
scandone
STATION_IDLE
add 0
aid 16
pm open phy_2,type:2 0 0
cnt 

connected with DHOME, channel 11
dhcp client start...
STATION_IDLE
ip:192.168.1.42,mask:255.255.255.0,gw:192.168.1.1
TCP: Connect to ip  192.168.1.128:1880
MQTT: Connected to broker 192.168.1.128:1880
MQTT: Sending, type: 1, id: 0000
TCP: Sent
TCP: Disconnected callback
MQTT: Disconnected
Free memory

PuceBaboon, 2015-06-18:
You might want to check the Mosquitto log on the Pi to see what reason the server gives for the disconnect.

harryd100, 2015-06-18:
Without the source code it's hard to tell what happens or should happen!!

tuanpmt, 2015-06-18:
Maybe here: https://github.com/tuanpmt/esp_mqtt/issues/54

closed 2016-09-09 by tuanpmt