tuanpmt/esp_mqtt #66: Compiling without any issues, but it won't run at all

bonie89, 2015-07-06:
I haven't changed anything in the source code. It is compiling without any issues and I’m also able to flash without a problem. But I think the controller won't even boot. There are no signs that anything is running except for very little output over the serial port when powered up (mostly nulls).
I tried to comment things out in user_main.c until it started to run. I had to comment out everything that was related to mqtt... then it connected to my wifi (specified in user_config.h) and I was also only then able to see any output over the serial port. 
I did the changes to user_config.h only to STA_SSID and STA_PASS and also after I tried everything without any changes.

Can someone please direct me in a direction on how to solve this issue? 
Thank you in advance! And also a big thank you for investing so much time in such a project and making it open source!

bonie89, 2015-07-06:
Messing around a whole day with this...also reading other issues, but not the right one #62.
So I changed in the makefile:
```
flash: $(FW_FILE_1)  $(FW_FILE_2)
	$(ESPTOOL) -p $(ESPPORT) write_flash $(FW_1) $(FW_FILE_1) $(FW_2) $(FW_FILE_2)
```
to:
```
flash: $(FW_FILE_1)  $(FW_FILE_2)
	$(ESPTOOL) -p $(ESPPORT) write_flash $(FW_2) $(FW_FILE_2) $(FW_1) $(FW_FILE_1)
```
then everything worked fine

closed 2015-07-06 by bonie89

LongHairedHacker, 2015-08-27:
Just ran into the same problem.
Found this issue after about 2h looking for the error in my sources ... thanks a lot for the tip.
Saved my evening.