tuanpmt/esp_mqtt #67: Sending keepalive during submission breaks the queue

Sermus, 2015-07-15:
I very often see the following pattern in the logs:
MQTT: queuing publish, length: 47, queue size(0/2048)
MQTT: queuing publish, length: 44, queue size(49/2048)
MQTT: queuing publish, length: 46, queue size(95/2048)
MQTT: queuing publish, length: 42, queue size(143/2048)
MQTT: queuing publish, length: 38, queue size(187/2048)
MQTT: queuing publish, length: 110, queue size(227/2048)
MQTT: queuing publish, length: 496, queue size(339/2048)
MQTT: queuing publish, length: 44, queue size(837/2048)
MQTT: queuing publish, length: 43, queue size(883/2048)
MQTT: queuing publish, length: 46, queue size(928/2048)
MQTT: queuing publish, length: 40, queue size(976/2048)
MQTT: queuing publish, length: 46, queue size(1018/2048)
MQTT: queuing publish, length: 50, queue size(1066/2048)
MQTT: queuing publish, length: 472, queue size(1118/2048)
MQTT: queuing publish, length: 32, queue size(1592/2048)
MQTT: Sending, type: 3, id: 0000
TCP: Sent
MQTT: Sending, type: 3, id: 0000
TCP: Sent
MQTT: Sending, type: 3, id: 0000
TCP: Sent
MQTT: Sending, type: 3, id: 0000
TCP: Sent
MQTT: Sending, type: 3, id: 0000
TCP: Sent
MQTT: Sending, type: 3, id: 0000
TCP: Sent
MQTT: Sending, type: 3, id: 0000
TCP: Sent
MQTT: Sending, type: 3, id: 0000
TCP: Sent
MQTT: Sending, type: 3, id: 0000

**MQTT: Send keepalive packet to XX.XX.XX.XX:xxx!
MQTT: Sending, type: 12, id: 0000**
TCP: data received 2 bytes
TCP: data received 18 bytes
MQTT: queuing publish, length: 47, queue size(698/2048)
MQTT: queuing publish, length: 44, queue size(747/2048)
MQTT: queuing publish, length: 46, queue size(793/2048)
MQTT: queuing publish, length: 42, queue size(841/2048)
MQTT: queuing publish, length: 38, queue size(885/2048)
MQTT: queuing publish, length: 110, queue size(925/2048)
MQTT: queuing publish, length: 496, queue size(1037/2048)
MQTT: queuing publish, length: 44, queue size(1535/2048)
MQTT: queuing publish, length: 43, queue size(1581/2048)
MQTT: queuing publish, length: 46, queue size(1626/2048)
MQTT: queuing publish, length: 40, queue size(1674/2048)
MQTT: queuing publish, length: 46, queue size(1716/2048)
MQTT: queuing publish, length: 50, queue size(1764/2048)
MQTT: queuing publish, length: 472, queue size(1816/2048)
MQTT: Queue full
MQTT: Queue full
MQTT: Queue full

After that queue size never goes down.

Sermus, 2015-07-15:
It seems that sending a keepalive packet at the middle of submitting messages breaks the queue

quant67, 2015-11-20:
I had the some problem. and change espconn.*_sent() to espconn.*_send(), although did not known why, but it seems work.

commit 3e2c77de (2016-02-17): Do not send keep-alive before pending send has finished

According to the SDK documentation espconn_send should not be called
before espconn_sent_callback has been invoked for the previous
call.

Fixes #67
closed 2016-02-17 by tuanpmt