tuanpmt/esp_mqtt #68: Compiling with ESP8266 SDK (esp_iot_sdk_v1.2.0_15_07_03) gives error

viktak, 2015-07-22:
Compiling the project with the latest SDK (1.2.0) gives the following error:
```
CC driver/uart.c
In file included from driver/uart.c:13:0:
/opt/Espressif/esp_iot_sdk_v1.2.0/include/osapi.h:8:20: fatal error: string.h: No such file or directory
 #include <string.h>
                    ^
compilation terminated.
Makefile:216: recipe for target 'build/driver/uart.o' failed
make: *** [build/driver/uart.o] Error 1
```
I can't find string.h anywhere in the SDK. Has the development environment changed?

If I add the missing files to the project (from another SDK), the build still fails:
'''
CC driver/uart.c
CC mqtt/mqtt.c
CC mqtt/ringbuf.c
CC mqtt/mqtt_msg.c
CC mqtt/queue.c
CC mqtt/utils.c
CC mqtt/proto.c
CC user/onewire.c
CC user/user_main.c
CC modules/config.c
CC modules/wifi.c
AR build/app_app.a
LD build/app.out
/opt/Espressif/crosstool-NG/builds/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot find -lc
/opt/Espressif/crosstool-NG/builds/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot find -lhal
collect2: error: ld returned 1 exit status
Makefile:184: recipe for target 'build/app.out' failed
make: *** [build/app.out] Error 1
'''

Has the toolchain changed?

McGr3g0r, 2015-07-22:
Hi,
You probably forgot to get INCLUDE files https://github.com/esp8266/esp8266-wiki/blob/master/include.tgz when setting new SDK workspace folder, same thing about libc and libhal (https://github.com/esp8266/esp8266-wiki/tree/master/libs).

viktak, 2015-07-22:
Yes, the include.tgz was indeed missing, but I still get the error at the end:

'''
C driver/uart.c
CC mqtt/mqtt.c
CC mqtt/ringbuf.c
CC mqtt/mqtt_msg.c
CC mqtt/queue.c
CC mqtt/utils.c
CC mqtt/proto.c
CC user/onewire.c
CC user/user_main.c
CC modules/config.c
CC modules/wifi.c
AR build/app_app.a
LD build/app.out
/opt/Espressif/crosstool-NG/builds/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot find -lc
/opt/Espressif/crosstool-NG/builds/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot find -lhal
collect2: error: ld returned 1 exit status
Makefile:184: recipe for target 'build/app.out' failed
make: *** [build/app.out] Error 1
'''

I'm using the same, unchanged virtual machine as for the previous SDKs.

McGr3g0r, 2015-07-22:
You seem no to have libhal.a and libc.a libraries in SDK lib folder - you can download it from here https://github.com/esp8266/esp8266-wiki/tree/master/libs

viktak, 2015-07-23:
Finally, I understood this:
I first added libhal.a and libc.a to the project's include files -> got errors.
Then I added them to the SDK's include files -> got errors.
Finally, I added them to the SDK's lib files -> now it compiles/builds/runs with no problem!

Thanks for the help!!!

Is there any reason these files are not in the SDK by default? Also, I haven't found anything on them in the SDK's documentation. Was I sloppy reading the instructions or it's missing indeed?

This issue can now be closed. Thanks for your assistance!

sreeisalso, 2015-09-06:
Please close the issues since it is mot code issue any more

viktak, 2015-09-08:
Issue closed as per above.

closed 2015-09-08 by viktak

Rajarajan2809, 2017-09-14:
Thank you @viktak @McGr3g0r

bha14ur, 2019-01-23:
when i trying to compile blinky example using SDK on VM under "https://www.youtube.com/watch?v=58bdrN0pkbc" video guide line and enter below code
/Share/esp_iot_sdk_v1.2.0_15_07_03/esp_iot_sdk_v1.2.0/app$    make

i get these warnings and error:

../Makefile:300: warning: overriding commands for target `clean'
Makefile:300: warning: ignoring old commands for target `clean'
../Makefile:304: warning: overriding commands for target `clobber'
Makefile:304: warning: ignoring old commands for target `clobber'
../Makefile:308: warning: overriding commands for target `.subdirs'
Makefile:308: warning: ignoring old commands for target `.subdirs'
make[1]: Entering directory `/mnt/Share/esp_iot_sdk_v1.2.0_15_07_03/esp_iot_sdk_v1.2.0/app/user'
../../Makefile:300: warning: overriding commands for target `clean'
../Makefile:300: warning: ignoring old commands for target `clean'
../../Makefile:304: warning: overriding commands for target `clobber'
../Makefile:304: warning: ignoring old commands for target `clobber'
../../Makefile:308: warning: overriding commands for target `.subdirs'
../Makefile:308: warning: ignoring old commands for target `.subdirs'
../../Makefile:361: warning: overriding commands for target `.output///lib/libuser.a'
../Makefile:361: warning: ignoring old commands for target `.output///lib/libuser.a'
make[1]: Leaving directory `/mnt/Share/esp_iot_sdk_v1.2.0_15_07_03/esp_iot_sdk_v1.2.0/app/user'

sweetpants, 2019-01-23:
You are using an ancient SDK version. Try using v2.x version first

bha14ur, 2019-01-23:
I am not finding any v2.x version in https://github.com/esp8266/esp8266-wiki/tree/master/sdk

sweetpants, 2019-01-23:
I think this site is not maintained anymore (4 years old). Try https://www.espressif.com/en/support/download/sdks-demos