tuanpmt/esp_mqtt #73: without usrname and passsword don't support.

leeshineSZ, 2015-09-10:
without username and password don't support. Can provide a interface which has a flag  to set no username and password.

zack-wang, 2015-09-10:
its server side problem? you can make server ignore your username and password.

Namphibian, 2015-09-10:
You can connect to the eclipse MQTT(mosquito server) at m2m.eclipse.org with a user name and password and the server ignores it.

leeshineSZ, 2015-09-10:
Thank you for you answer, I mean when my server don't need usrname and password, i can't ingore it in the client. And it will increase the size of the package. I want it less bandwidth consuming.  

lichgu@aliyun.com
 
From: Neil
Date: 2015-09-10 12:25
To: tuanpmt/esp_mqtt
CC: leeshineSZ
Subject: Re: [esp_mqtt] without usrname and passsword don't support. (#73)
You can connect to the eclipse MQTT(mosquito server) at m2m.eclipse.org with a user name and password and the server ignores it.

zack-wang, 2015-09-10:
In real world, you'll need this authentication for safety reason. Without password, anyone can login to your server and "subscribe" any topic! btw, the username can be a character "u" and password can be "p" to login your broker. And this username and password only for make "connection" they only use once! they won't cost any bandwidth in regular message exchange. Are you mean the size of code binary?

leeshineSZ, 2015-09-13:
ok, thankyou!

closed 2015-09-13 by leeshineSZ