tuanpmt/esp_mqtt #76: Stable?

aignacio, 2015-09-25:
Hello @tuanpmt  I would like to know how stable is this version of this port made by you, did you test a network with a large number of devices? Is there some problem of ESP die and not reboot again? I'm thinking in expand your lib to 500 ESPs and a large network MQTT and if you can answer some of this questions I will sleep better. =)Thanks

zack-wang, 2015-09-28:
I've one machine running for 3 months. Outdoor and signal level about -50dBm. Never die or reboot... Another one for 1.5months,even in high temperature, and packed in a open iron box(i have to hide this device in customer's mailbox)

tuanpmt, 2016-09-09:
@aignacio 1 year has passed, how about your devices?

aignacio, 2016-09-09:
Hello @tuanpmt I'm using the old version of your port and it's incredible since they been working, none of them halted or had problem in execution. Great Job, thanks! Now I'm working on 6LoWPAN devices using Contiki, I'm porting MQTT-SN to this RTOS....

tuanpmt, 2016-09-09:
Nice, this really is a news indicating the reliability of esp8266,
Thanks

closed 2016-09-09 by tuanpmt