tuanpmt/esp_mqtt #78: MQTT routing

tomikpik, 2015-11-01:
Hi, I would like to know if mqtt on esp8266 would work even if I want to run mqtt on first esp8266 which is connected to second esp8266 which is then connected to the network where the mqtt broker runs.
Thank you.

Namphibian, 2015-11-01:
If the second ESP-8266 is only acting as a AP/Wifi Access Point yes this is theoretically feasible. However keep in mind this will probably affect your systems ability to scale as the ESP-8266 while a wonder of technology is still a small devices limited in memory. Thus it might work for 1,2,3,4 or even 20 MQTT clients pushing through data but at some point this one access point will be overloaded I suspect.

Maybe try a Raspberry Pie or Netduino as an access point. It wont break the bank and the units are considerably more powerful.

tomikpik, 2015-11-01:
I would like to create mesh network using ESP8266 and communicate via MQTT, so the question is if the ESP8266 can route the MQTT packets to broker from other ESPs that are connected to the first ESP. 
And I've read somewhere that maximum number of clients connected to ESP AP is 4. Is that true?

Namphibian, 2015-11-01:
The ESP in AP mode is a full AP thus yes it can route any TCP/IP or UDP traffic. The question is not if it can the question is more along the lines of can the ESP acting as a AP handle scaling of systems.