tuanpmt/esp_mqtt #80: Error In function `wifi_check_ip'

saytinh, 2015-12-15:
LD build/app.out
build/app_app.a(wifi.o): In function `wifi_check_ip':
/home/saytinh/esp_mqtt/modules/wifi.c:57: undefined reference to `wifi_set_opmode_current'
/home/saytinh/esp_mqtt/modules/wifi.c:57: undefined reference to `wifi_station_set_config_current'
/home/saytinh/esp_mqtt/modules/wifi.c:66: undefined reference to `wifi_set_opmode_current'
/home/saytinh/esp_mqtt/modules/wifi.c:74: undefined reference to `wifi_station_set_config_current'
collect2: error: ld returned 1 exit status
make: *** [build/app.out] Error 1

Dear all Guys,

I compile and received an error like above. Anyone can help me to resolve this issue. Thank so much

harryd100, 2015-12-15:
Are you using the latest esp_mqtt from github? I'm using this library for a long time now and don't see this error. Did you add #include "wifi.h" to your user_main.c?

saytinh, 2015-12-16:
Thanks harryd100.

I resolved my issue. it's because not completed toolchain setup.

closed 2015-12-16 by saytinh