tuanpmt/esp_mqtt #81: Receive one only retained message after multiple subscribes

Suxsem, 2015-12-15:
Hi. I'm going crazy with this problem:

in the onConnected callback I have multiple subscriptions (4 of them) with qos=2.
All of the 4 topics have retained messages.

The problem is that I only get a onMessage callback for the first topic.

If I send new messages to those topics I receive all of them. The problem is related only to the retained messages that should be received after subscribe; as I said i only receive the retained message related to the first subscribed topic.

Please take a look, it's really important. thanks!

Suxsem, 2015-12-15:
@tuanpmt help :D

aignacio, 2016-02-10:
help!!!

sconwayaus, 2016-04-01:
I have the same problem, albeit I'm working off an older code base but I think the code hasn't changed that much.

The problem seems to be that a single TCP packet from my mosquito server contains 3 messages in the following order, PUBHLISH, SUBACK, PUBLISH. The first two get processed fine, but the third gets discarded. Looking at mqtt_tcpclient_recv(), it only loops back if the last message was a PUBLISH, so commenting out this check resolves the issue for me:

```
			// NOTE: this is done down here and not in the switch case above
			// because the PSOCK_READBUF_LEN() won't work inside a switch
			// statement due to the way protothreads resume.
			//*** COMMENT OUT THIS IF STATEMENT ***
//			if(msg_type == MQTT_MSG_TYPE_PUBLISH)
//			{
			  len = client->mqtt_state.message_length_read;

			  if(client->mqtt_state.message_length < client->mqtt_state.message_length_read)
			  {
				  //client->connState = MQTT_PUBLISH_RECV;
				  //Not Implement yet
				  len -= client->mqtt_state.message_length;
				  pdata += client->mqtt_state.message_length;

				  INFO("Get another published message\r\n");
				  goto READPACKET;
			  }

//			}
```

harryd100, 2016-10-24:
I have the same issue on multiple subscribes. I only receive the retained message on the first topic. Going to try the above mentioned solution soon

harryd100, 2016-10-24:
Already did the test, and it seems to work fine now. Both topics receive the retained messages now. I'm running the latest version of MQTT and SDK_2.0.0
@sconwayaus, you made my day, thanks a lot