tuanpmt/esp_mqtt #82: subscribes qos is ignored

Suxsem, 2015-12-15:
qos is not used in MQTT_Subscribe(MQTT_Client *client, char* topic, uint8_t qos)

all subscribes are done with qos=0

client->mqtt_state.outbound_message = mqtt_msg_subscribe(&client->mqtt_state.mqtt_connection,
											topic, 0,
											&client->mqtt_state.pending_msg_id);

must be fixed as soon as possible

harryd100, 2015-12-16:
"must be fixed as soon as possible", this is an Open Source project where people put in a lot of free time to make something free of costs. If you ask nicely, maybe Tuan is willing to fix this for all of us if this is a bug.

Namphibian, 2015-12-16:
Agree with harryd100 here.

Ask nicely or better yet as the code is available why dont you give a
helping hand and fix the code? Help the project out if you are using it. I
have 101 outstanding items on this library but since it is not my full time
job I have to find time to get things done.

Talking of which I need to do a pull request soon I have some mods that
needs to go in but they are all Wifi and REST related.

closed 2016-02-02 by tuanpmt