tuanpmt/esp_mqtt #83: MQTT bridge that utilizes the services of the MQTT server and MQTT client running in parallel ...is it possible with esp_mqtt with modification?

hardikharpal, 2015-12-18:
Hi tuanpmt ,
Is it possible to build client server (MQTT bridge ) which can serve following purpose explained bellow on esp platform? 
This application implements a MQTT bridge that utilizes the services of the MQTT server and MQTT client running in parallel (on the same ESP8266). The MQTT server manages the local MQTT network whereas, the MQTT client connects to a cloud server / broker. The bridge application enables data exchange between local-server and the cloud server. To receive data from the cloud server, the MQTT bridge, through the MQTT client, subscribes to certain topics. It also forwards the data / topics received from the local MQTT server to the cloud-server. The local MQTT server handles the connections from the multiple MQTT clients (also shown as Launchpads in the following figure, but can be running on another platform) in the local network. 
http://processors.wiki.ti.com/index.php/CC32xx_MQTT_Server

Namphibian, 2015-12-18:
MQTT is peer to peer there are no clients and there are no servers.
Implementing a full MQTT broker on a ESP devices is going to be hard with
the limitations in memory for example as a matter of fact I don’t think it
is realistic with the current devices. Maybe with the ESP32

Namphibian, 2015-12-18:
Sorry I pressed send too quickly wanted to edit that mail. So MQTT is not a
traditional client server model. Client connect to a broker and subscribe
to topics. If you could perhaps use these terms in your question it might
be clearer.

hardikharpal, 2015-12-18:
Hi Namphibian,
Thank you for your response.
Sorry for my improper description.
Use case explained in http://processors.wiki.ti.com/index.php/CC32xx_MQTT_Server  is what I want to implement.
It is possible with TI cc3200 right now.

tuanpmt, 2016-09-09:
It is possible to run a small broker on ESP8266. But I do not see any benefit for implements it