tuanpmt/esp_mqtt #86: MQTT publish with timer

adityaalviori, 2015-12-23:
I am confuse when publish data to server with 10sec delay.
For example, i want to publish a temperature data every 10sec

If possible using timer??

i use mqtt in eclipse IDE 

thanks

harryd100, 2015-12-23:
Yes this is possible,

#define ALIVE_TIME				10000
static volatile os_timer_t alive_timer;

os_timer_disarm(&alive_timer);												
os_timer_setfn(&alive_timer, (os_timer_func_t *)alive_timerfunc, NULL);

void ICACHE_FLASH_ATTR mqttConnectedCb(uint32_t *args)
{
  // arm your timer
  os_timer_arm(&alive_timer, ALIVE_TIME, 1);
}
void ICACHE_FLASH_ATTR mqttDisconnectedCb(uint32_t *args)
{
  // When disconnected
  os_timer_disarm(&alive_timer);
}

void ICACHE_FLASH_ATTR alive_timerfunc(uint32_t *args)
{
  // do your stuff here
}

adityaalviori, 2015-12-26:
thanks for your help harryd100
after i tried your program, i have some question
1.  os_timer_disarm(&alive_timer);
     os_timer_setfn(&alive_timer, (os_timer_func_t *)alive_timerfunc, NULL);

harryd100, 2015-12-28:
Yes,
  os_timer_disarm(&alive_timer);
  os_timer_setfn(&alive_timer, (os_timer_func_t *)alive_timerfunc, NULL );
should be in user_init()
I use this to publish power usage and voltage to my broker every 10 seconds
You also have to setup the other functions, this alone will not work.
Arm the timer after connecting to the MQTT broker (mqttConnectedC), disarm when disconnected from MQTT broker (mqttDisconnectedCb) . Your (alive_timerfunc) handles the publishing of the data to the broker.

adityaalviori, 2015-12-29:
thanks for your replies harryd100
your program to get power usage data and voltage in (alive_timerfunc)?
my (alive_timerfunc) 
void ICACHE_FLASH_ATTR alive_timerfunc(uint32_t *args)
{
	MQTT_Client* client = (MQTT_Client*)args;
	MQTT_Publish(client, "/pub/0", "hello0", 6, 0, 0);
}

is it right??
void ICACHE_FLASH_ATTR alive_timerfunc(uint32_t *args)
{
	MQTT_Client* client = (MQTT_Client*)args;
	MQTT_Publish(client, "/pub/0", "hello0", 6, 0, 0);
} 
when i use this function, the mqtt didn't publish
i have one question, when disarm, the device disconnected to broker? so disarm make the device disconnected to broker?? 

if i use your program to subscribe, the device cannot subscribe all of data

thnks for your help

closed 2018-04-12 by adityaalviori