tuanpmt/esp_mqtt #87: MQTT_Disconnect don't disconnect

sifourquier, 2015-12-27:
Hi
MQTT_Disconnect clear memory but don't close TCP connection

I have edit the mqtt.c for close the TCP but if you can check

Tanks you for your work
[mqtt.c.zip](https://github.com/tuanpmt/esp_mqtt/files/72693/mqtt.c.zip)

luuke, 2015-12-28:
Hey
With old version my ESP-07 module was having exception watchdog resets after disconnecting from MQTT and (straight after that) disconnecting from WIFI. Module was reseting in about 95% attempts.

Just replaced mqtt.c file with your version and so far my module is working fine as it is supposed to.
Was trying to find the problem for few days already. You made my day!

Thanks,
Luuke

sifourquier, 2015-12-28:
Hi think i have found a "error in me new code"
of MQTT are disconnected for another reason of a desired disconnect the esp don't reconnect.
If i found a best solution i post.

vowstar, 2016-01-01:
@sifourquier You can try this https://github.com/vowstar/esp_mqtt, I'm testing and it is OK to safe disconnect MQTT client.
The whole project is still under testing so I'm not sure to create pull request to upstream, until it is stable.
You can use MQTT_Disconnect(MQTT_Client *mqttClient) to disconnect TCP connection,
and use MQTT_DeleteClient(MQTT_Client *mqttClient) to fully free the memory. When you delete client, you must redo MQTT_InitClient. When you use MQTT_Disconnect, need'nt re init client.

Cicero-MF, 2016-05-27:
To me, MQTT_Disconnect() should send a mqtt_msg_disconnect() message, message type  MQTT_MSG_TYPE_DISCONNECT = 14.  But that function isn't used.