tuanpmt/esp_mqtt #88: Step-by-step using SSL with cert, Key and CA [Newbiew here]

silverfangs, 2016-01-11:
Hi guys,

I know Im asking very silly question, but I'm still newbie try to implement this SSL in my mqtt connection. Can you guys show me a step-by-step way to get it done with using SSL cert, key and CA file (if there is any)?  I have tried to do my own search and study the code, but I felt very hopeless cant understand fully how it works.

If any of you guys done it success, can show me some examples? Kindly appreciate for the effort!

Deadolus, 2016-01-21:
I also have problems with this. 
I also created a post on the official ESP8266 espressif forum: http://bbs.espressif.com/viewtopic.php?f=7&t=1639&p=5472#p5472

quant67, 2016-01-21:
Finally, I made it several hours ago.

I used esp_mqtt as a library so want it as a client.

- Firstly, I asked the server engineer to give me a CA file, then he gave me a file name cacert.pem.

- Secondly, converted CA file to cer format. At my linux, I installed openssl then enter the command bellow:
	openssl x509 -inform pem -in cacert.pem -out cacert.cer

- Thirdly, converted cer file to esp_ca_cert.bin. You may want python script I used bellow. Just run it at the same directory as cacert.cer:
```
import os

class Cert(object):
    def __init__(self, name, buff):
        self.name = name
        self.len = len(buff)
        self.buff = buff
        pass

    def __str__(self):
        out_str = ['\0']*32
        for i in range(len(self.name)):
            out_str[i] = self.name[i]
        out_str = "".join(out_str)
        out_str += str(chr(self.len & 0xFF))
        out_str += str(chr((self.len & 0xFF00) >> 8))
        out_str += self.buff
        return out_str
        pass

def main():
    cert_list = []
    file_list = os.listdir(os.getcwd())
    cert_file_list = []
    for _file in file_list:
        pos = _file.find(".cer")
        if pos != -1:
            cert_file_list.append(_file[:pos])

    for cert_file in cert_file_list:
        with open(cert_file+".cer", 'rb') as f:
            buff = f.read()
        cert_list.append(Cert(cert_file, buff))
    with open('esp_ca_cert.bin', 'wb+') as f:
        for _cert in cert_list:
            f.write("%s" % _cert)
    pass
if __name__ == '__main__':
    main()
```

- Fourthly, added a line of code to user_init function of my esp8266 program:
```
	espconn_secure_ca_enable(0x01, 0x7c);
```
The argument 0x01 means that I want program at as an ssl client, and 0x7c means that I will burn esp_ca_cert.bin into 0x7c000.

- Finally, burn esp_ca_cert.bin into 0x7c000.

Now I can write a program that connect with my server using mqtt with ssl.

May the luck be with you.

Spiritvn, 2016-01-21:
@quant67 thanks, it's very detail.

Deadolus, 2016-01-21:
Yes, thank you @quant67 this did also help me!
I now get "This file IS a PEM file" which is good. I now also no longer receive an empty certificate on the server. 
Still no luck connecting to the Amazon AWS cloud. 
Might be a bad certificate which I flashed in to the chip, maybe something is still missing. 
I created a new post at http://bbs.espressif.com/viewtopic.php?f=7&t=1639&p=5472#p5472, after incorporating your changes and detailing where I am stuck now.

vidia, 2016-02-12:
@Deadolus Are you trying to connect to AWS IoT specifically? I am working on that as well, have you made any progress?

Deadolus, 2016-02-12:
@vidia It does not seem to work at the momen as Esp does not support TLS 1.2.
Read about my odysee athttp://bbs.espressif.com/viewtopic.php?f=7&t=1639&p=5522&hilit=Aws+iot#p5522

TomaszKn, 2016-03-18:
@quant67 It works for you?
For my isn't work.

quant67, 2016-03-18:
Of cource it work at my server. I use rabbitmq in LVS as the MQTT broker.
But never try aws.

AcuarioCat, 2016-04-02:
See my post here:
https://github.com/tuanpmt/esp_mqtt/issues/98