tuanpmt/esp_mqtt #96: Disconnect before reconnecting

harryd100, 2016-03-05:
Hi, Since last change "Disconnect before reconnecting" I have an issue when my ESP roams to another AP. Although Wifi comes up and I have an IP address, MQTT is not working anymore. After a reset everything works again. After removing the changes from "Disconnect before reconnecting" it works again, also when the AP roams to another AP.

lorenzph, 2016-03-22:
Do you disconnect the previous connection (via MQTT_Disconnect) after the connection to the other API is established?

This change dropped the MQTT_Disconnect call which was happening inside MQTT_Connect if the connection was already established. The call was dropped as disconnection itself is a multi-step process:

1. Call to espconn_disconnect
2. Wait for mqtt_tcpclient_discon_cb and act according to the current state

If the connection is disconnected without waiting for the invocation mqtt_tcpclient_discon_cb, the disconnection callback may be invoked at any point in time. Depending on the current state this may lead to spurious reconnects.

harryd100, 2016-03-22:
Hi, I'm pretty much use the same code as in the library example, so i don't know what you mean with "Do you disconnect the previous connection (via MQTT_Disconnect) after the connection to the other API is established?". Since removing the "Disconnect before reconnecting" code-change all my devices work flawlessly for weeks now

lorenzph, 2016-03-22:
I guess you're referring to https://github.com/tuanpmt/esp_mqtt/blob/master/user/user_main.c? If that's the case can you try if adding a MQTT_Disconnect(&mqttClient); before the call to MQTT_Connect(&mqttClient) in line 46 mitigates the issue?

I'll try to reproduce this locally in the next couple of days.

closed 2016-04-04 by tuanpmt

harryd100, 2016-04-04:
Hi, Just tested this new commit and seems to work much better. Roaming between AP's no reconnects the ESP again. Thanks for fixing.

PR #101 (2016-04-04): Delete existing connection before connecting
If the current connection is terminated unexpectedly (e.g. because the
WiFi connection is dropped) the disconnect callback is not called. This
means that reestablishing the existing connection is not possible any
more.

Avoid this issue by forcefully removing the existing connection before
initiating the reconnection. Calling MQTT_Disconnect and waiting for
the disconnection callback should still be performed for intentional
disconnections.

Fixes #96
commit 0f980bf7 (2016-04-04): Delete existing connection before connecting

If the current connection is terminated unexpectedly (e.g. because the
WiFi connection is dropped) the disconnect callback is not called. This
means that reestablishing the existing connection is not possible any
more.

Avoid this issue by forcefully removing the existing connection before
initiating the reconnection. Calling MQTT_Disconnect and waiting for
the disconnection callback should still be performed for intentional
disconnections.

Fixes #96