tuanpmt/esp_mqtt #97: about

baoblackcoal, 2016-03-20:
Hi, 
i have got a issue, the logs:

MQTT: Sending, type: 12, id: 0000
mqtt_send_keepalive tcp send result=-7
TCP: Disconnected callback
MQTT: Disconnected

it has been very frequently, and LWT mssage is easy triggered. 
for fixing this issue i would be give 3 times chance to send keepalive,  is that OK?

baoblackcoal, 2016-03-21:
i think i can use the MQTT_SEND_TIMOUT instead of 3

vowstar, 2016-09-06:
I got same issue, due to very bad network and special mqtt server configuration. So I changed keepalive interval to half of keepalive, because "If a client does not receive a PINGRESP message within a Keep Alive time period after sending a PINGREQ, it should close the TCP/IP socket connection." in [MQTT 3.1 document](http://public.dhe.ibm.com/software/dw/webservices/ws-mqtt/MQTT_V3.1_Protocol_Specific.pdf), I found that half keepalive will stable.

PR #118 (2016-09-06): Some fix about stability of mqtt connection about keepalive
Close #97 
I got same issue (using default setting on [emqtt server](https://github.com/emqtt/emqttd)), due to very bad network and special mqtt server configuration. So I changed keepalive interval to half of keepalive, because "If a client does not receive a PINGRESP message within a Keep Alive time period after sending a PINGREQ, it should close the TCP/IP socket connection." in [MQTT 3.1 document](http://public.dhe.ibm.com/software/dw/webservices/ws-mqtt/MQTT_V3.1_Protocol_Specific.pdf), I found that half keepalive will stable.

* [... cut]
closed 2016-09-06 by tuanpmt