tuanpmt/esp_mqtt #98: Config Mosquitto broker with SSL

TomaszKn, 2016-03-30:
Hi
I Create SSL Self sign:

> openssl req -x509 -newkey rsa:1024 -keyout key.pem -out cert.pem -days 100

Anyone know how to configure Mosquitto broker to work with ESP8266 over SSL?

Thanks
Tom

AcuarioCat, 2016-04-02:
Hi,
you need to create certificates and copy them to the correct directory. The other crucial thing is that when you create the certificate using OpenSSL it forces you to input a pass phrase. Mosquitto won't work with a pass phrase so you need to remove it from the certificate.

Also ESP doesn't support TLS 1.2 properly, it will attempt to connect then fail so you need to force tlsv1 in the mosquito config file.

The following is for Raspberry PI
 
Set up Mosquitto TLS
 ==========================================
 (Create a directory to work in then...)
 openssl req -x509 -newkey rsa:1024 -keyout ca.crt -out cert.crt -days XXX
(12345 for pass phrase and enters or if you want add CA information)
openssl rsa -in ca.crt -out newca.pem
(12345 for pass phrase)

cp files to dest /etc/mosquitto/certs/

Modify Mosquitto config file /etc/mosquitto/mosquitto.conf
# mosquitto.conf
cafile /etc/mosquitto/certs/cert.crt
certfile /etc/mosquitto/certs/cert.crt
keyfile /etc/mosquitto/certs/newca.pem
require_certificate false
tls_version tlsv1

Hope this helps!

TomaszKn, 2016-04-03:
Thanks for answer.
But I need have cacert.cer file to create esp_ca_cert.bin.
cert.crt is this file ?

AcuarioCat, 2016-04-03:
Yes, the two are the same. If you follow exactly above then it should work for you. This is exactly what I have done and it works for me. I've set up 3 systems the same way and all are working. 

You don't need to create an esp_ca_cert.bin file as the broker connects with SSL using unauthenticated encrypted support (the same as a web browser does to an SSL website).

When you call MQTT_InitConnection call it with security = 1 to enable SSL connections the client should then connect to your broker.

julliermedias, 2016-05-09:
Thank you, @AcuarioCat, for the instructions.
I'm using Mosquitto Linux broker and 1.3.0 SDK in my esp. Without SSL, it works fine, but with SSL and following all your steps, I have no sucess. Is there any step not explicited? Can you tell me what is your SDK version?
In debugger mode, the application hangs in "TCP: connecting...", something like that:

> System started ...
> STATION_IDLE
> STATION_IDLE
> STATION_IDLE
> STATION_IDLE
> WIFI CONNECTED
> STATION_IDLE
> WIFI GOT IP
> TCP: Connect to domain domain.com:portnumber
> DNS: found ip XXX.XXX.XXX.XXX
> TCP: connecting...
> 

I cannot change my SDK, because I have other applications using it.
One more time, thank you very much.

harryd100, 2016-05-09:
You can install multiple SDK's and modify your Makefile to use the SDK you want/need.

tuanpmt, 2016-09-09:
SDK 2.0 with mbedtls work with SSL smooth

closed 2016-09-09 by tuanpmt

BalrogMithrandir, 2017-05-05:
Hi @AcuarioCat! I am using esp8266 as client to connect with mosquitto through ssl. You said that we can disable authentication of ssl,  I tried it and it was done. But now I want to enable one-way certification, that is to say I want to make esp8266 verify the mosquitto broker. I want to know can i use .crt instead of .cer to make  esp_ca_cert.bin? thanks!