tuanpmt/esp_mqtt #102: Seems some problem with mqtt reconnect

sreeisalso, 2016-04-06:
Hi @tuanpmt 

There is some problem with mqtt reconnect I get the following log very often

`` DNS: Found, but got no ip, try to reconnect``
 ``TCP: Reconnect to 192.168.0.2:1883``

Namphibian, 2016-04-06:
Hi.

That appears DNS related. Looks like an internal IP address so maybe the
dns on your router is behaving bad.

Kind Regards
Neil Franken

sreeisalso, 2016-04-06:
Hi @Namphibian  

But after rebooting the esp it works perfectly fine.

Thanks
Sreekanth

Namphibian, 2016-04-06:
Hi Sreekanth

You mean like how windows magically works again when you reboot it? It
certainly appears to be a DNS error and DNS services on your network would
be a starting point to eliminate this.

It might be that DNS is timing out for example. I had similar issues when
using the eclipse project MQTT server during testing. My service provider
at Telstra had some DNS issues.

Just some wild speculation from my part as I have very little information
to go on. As a suggestion why not block dns queries for your esp mac
address and see if you get the dame error.

Kind Regards
Neil Franken

harryd100, 2016-04-06:
Take a look at https://github.com/tuanpmt/esp_mqtt/issues/96. Could it be the ESP disconnects from Wifi temporarily? Using the latest build should solve your problem.

sreeisalso, 2016-04-06:
@Namphibian It doesn't seem to be DNS issue, because it will work perfectly well if I reboot esp alone. without modifying anything in the network.

@harryd100 this doesn't seem to be related #96. anyhow I have taken #96 as well in the latest build.

sreeisalso, 2016-05-13:
Hi team,

This issue is not appearing after taking #96 

Thanks 
Sreekanth

closed 2016-05-13 by sreeisalso