tuanpmt/esp_mqtt #104: Cannot get started

CROSP, 2016-04-29:
Hi, thank you for your firmware.
I have just tried to compile and load firmware into my ESP 
But it won't work. 
I have followed all your instructions, everything compiles well.
Here is my user_main.c

```
#include "ets_sys.h"
#include "driver/uart.h"
#include "osapi.h"
#include "mqtt.h"
#include "wifi.h"
#include "config.h"
#include "debug.h"
#include "gpio.h"
#include "user_interface.h"
#include "mem.h"

MQTT_Client mqttClient;

void wifiConnectCb(uint8_t status)
{
	if(status == STATION_GOT_IP){
		MQTT_Connect(&mqttClient);
	} else {
		MQTT_Disconnect(&mqttClient);
	}
}
void mqttConnectedCb(uint32_t *args)
{
	MQTT_Client* client = (MQTT_Client*)args;
	INFO("MQTT: Connected\r\n");
	MQTT_Subscribe(client, "/wh/home/communication/intercom", 0);

	MQTT_Publish(client, "/wh/home/communication/intercom", "hello0", 6, 0, 0);
}

void mqttDisconnectedCb(uint32_t *args)
{
	MQTT_Client* client = (MQTT_Client*)args;
	INFO("MQTT: Disconnected\r\n");
}

void mqttPublishedCb(uint32_t *args)
{
	MQTT_Client* client = (MQTT_Client*)args;
	INFO("MQTT: Published\r\n");
}

void mqttDataCb(uint32_t *args, const char* topic, uint32_t topic_len, const char *data, uint32_t data_len)
{
	char *topicBuf = (char*)os_zalloc(topic_len+1),
			*dataBuf = (char*)os_zalloc(data_len+1);

	MQTT_Client* client = (MQTT_Client*)args;

	os_memcpy(topicBuf, topic, topic_len);
	topicBuf[topic_len] = 0;

	os_memcpy(dataBuf, data, data_len);
	dataBuf[data_len] = 0;

	INFO("Receive topic: %s, data: %s \r\n", topicBuf, dataBuf);
	os_free(topicBuf);
	os_free(dataBuf);
}

void user_init(void)
{
	uart_init(BIT_RATE_115200, BIT_RATE_115200);
	os_delay_us(1000000);

	CFG_Load();
	INFO("\r\rHELLO\r\n")
	MQTT_InitConnection(&mqttClient, sysCfg.mqtt_host, sysCfg.mqtt_port, sysCfg.security);
	//MQTT_InitConnection(&mqttClient, "192.168.11.122", 1880, 0);

	MQTT_InitClient(&mqttClient, sysCfg.device_id, sysCfg.mqtt_user, sysCfg.mqtt_pass, sysCfg.mqtt_keepalive, 1);
	//MQTT_InitClient(&mqttClient, "client_id", "user", "pass", 120, 1);

	MQTT_InitLWT(&mqttClient, "/lwt", "offline", 0, 0);
	MQTT_OnConnected(&mqttClient, mqttConnectedCb);
	MQTT_OnDisconnected(&mqttClient, mqttDisconnectedCb);
	MQTT_OnPublished(&mqttClient, mqttPublishedCb);
	MQTT_OnData(&mqttClient, mqttDataCb);

	WIFI_Connect(sysCfg.sta_ssid, sysCfg.sta_pwd, wifiConnectCb);

	INFO("\r\nSystem started ...\r\n");
}
```

I also modified `include/mqtt_config.h` file in following way 

#ifndef __MQTT_CONFIG_H__
#define __MQTT_CONFIG_H__

#define CFG_HOLDER	0x00FF55A4	/* Change this value to load default configurations */
#define CFG_LOCATION	0x3C	/* Please don't change or if you know what you doing */
#define MQTT_SSL_ENABLE

/*DEFAULT CONFIGURATIONS*/

#define MQTT_HOST			"mqtt.iot.crosp.net" //or "mqtt.yourdomain.com"
#define MQTT_PORT			1883
#define MQTT_BUF_SIZE		1024
#define MQTT_KEEPALIVE		120	 /*second*/

#define MQTT_CLIENT_ID		"COM_INTERCOM_%08X"
#define MQTT_USER			"home_intercom"
#define MQTT_PASS			"password"

#define STA_SSID "MY_AP_NAME"
#define STA_PASS "MY_AP_PASSWORD"
#define STA_TYPE AUTH_WPA2_PSK

#define MQTT_RECONNECT_TIMEOUT 	5	/*second*/

#define DEFAULT_SECURITY	0
#define QUEUE_BUFFER_SIZE		 		2048

#define PROTOCOL_NAMEv31	/*MQTT version 3.1 compatible with Mosquitto v0.15*/
//PROTOCOL_NAMEv311			/*MQTT version 3.11 compatible with https://eclipse.org/paho/clients/testing/*/

#endif // __MQTT_CONFIG_H__

But I am getting next serial output

```
scandone
no A700E found, reconnect after 1s
reconnect
f r0, scandone
no A700E found, reconnect after 1s
reconnect
f 0, scandone
no A700E found, reconnect after 1s
reconnect
```

I cannot figure out where does A700E come from, I tried to grep all file but it wasn't found.
Please help, what I am doing wrong.
Thanks

curiousmuch, 2016-05-02:
Did you modify the CFG_HOLDER?

BG2BKK, 2016-08-31:
Hello，I don't know what CFG_HOLDER is， and I don't know which one decide the value of CFG_HOLDER. I use nodemcu but it doesn't work

harryd100, 2016-08-31:
CFG_HOLDER is in your include/mqtt_config.h. Just change the value to something else and the new configuration will be saved

BG2BKK, 2016-08-31:
Hi, thank you for your reply.
I understand what the CFG_HOLDER means, and this value can be defined by myself. But I don't know how to calculat and get the proper value. When I use nodemcu with 4MB flash,  the CFG_HOLDER and CFG_LOCATION may be different from value of ESP12 with 1MB or 512KB flash。I don't know what the CFG_HOLDER and CFG_LOCATION actually mean, so could you tell me sth about them? Thank you :)

harryd100, 2016-08-31:
As said in the mqtt_config.h, #define CFG_LOCATION 0x3C /* Please don't change or if you know what you doing */ , just increment the hex value op the CFG_LOCATION. I use the mqtt library also on ESP01, EPS07 and ESP12E with the same settings.

BG2BKK, 2016-09-01:
Thank you for your reply!

tuanpmt, 2016-09-09:
Now you don't need CFG_HOLDER any more

closed 2016-09-09 by tuanpmt