tuanpmt/esp_mqtt #105: TCP: data received 1452 bytes

FOX61103, 2016-05-02:
Hi,thank you for your esp_mqtt
it works well,but there is a problem on mqtt receive:
when mqtt data > 1452 bytes,the device will reset!

##########################################
TCP: data received 1452 bytes
ets Jan  8 2013,rst cause:4, boot mode:(1,4)
wdt reset
##########################################

how can I fix it?
thank you!

someburner, 2016-05-22:
1460 is the TCP Maximum segment size in lwipopts.h. Are you using an lwip-based sdk build?

devlinhtran, 2016-09-01:
I tested about 50 module esp8266. When i sent a payload in MQTT service with different length i saw: about 50% module can send up to 960  bytes or more in one frame. 30% can only send 660 byte maximum and. 20% just send 220 bytes maximum. So do know what happen?

wateras, 2017-03-10:
TCP: data received 184 bytes,heaplen=24936
length=88,{"d":{"id":"65535","items":[],"number":1,"type":0},"t":"2016-11-25T09:41:53.798Z","m":1}
len2=68,len3=116
Get another published message
lost packet=68
TCP: data received 68 bytes,heaplen=24936
receive data----860----------
client->connState-----17
TCP: data received 860 bytes,heaplen=25448
packetlen=928
length=88,{"d":{"id":"65535","items":[],"number":1,"type":0},"t":"2016-11-25T09:41:53.798Z","m":1}
len2=812,len3=116
Get another published message
TCP: data received 812 bytes,heaplen=25448
length=88,{"d":{"id":"65535","items":[],"number":1,"type":0},"t":"2016-11-25T09:41:53.798Z","m":1}
len2=696,len3=116
Get another published message
TCP: data received 696 bytes,heaplen=25448
length=88,{"d":{"id":"65535","items":[],"number":1,"type":0},"t":"2016-11-25T09:41:53.798Z","m":1}
len2=580,len3=116
Get another published message
TCP: data received 580 bytes,heaplen=25448
length=88,{"d":{"id":"65535","items":[],"number":1,"type":0},"t":"2016-11-25T09:41:53.798Z","m":1}
len2=464,len3=116
Get another published message
TCP: data received 464 bytes,heaplen=25448
length=88,{"d":{"id":"65535","items":[],"number":1,"type":0},"t":"2016-11-25T09:41:53.798Z","m":1}
len2=348,len3=116
Get another published message
TCP: data received 348 bytes,heaplen=25448
length=88,{"d":{"id":"65535","items":[],"number":1,"type":0},"t":"2016-11-25T09:41:53.798Z","m":1}
len2=232,len3=116
Get another published message
TCP: data received 232 bytes,heaplen=25448
length=88,{"d":{"id":"65535","items":[],"number":1,"type":0},"t":"2016-11-25T09:41:53.798Z","m":1}
len2=116,len3=116
Get another published message
TCP: data received 116 bytes,heaplen=25448
length=88,{"d":{"id":"65535","items":[],"number":1,"type":0},"t":"2016-11-25T09:41:53.798Z","m":1}
MQTT:test client->sendTimeout=0
==================================================
The above is my test script to send 100 command test, the original MQTT will lead to accept module reset, mainly because each entry MQTT accept callback function to receive data length is different, some data need to be in the second will arrive, so we're going to packets of incomplete stitching on the MQTT parsing, can appear otherwise module reset.

 i have update mqtt.c ,list:
https://github.com/wateras/esp_mqtt

tuanpmt, 2017-03-11:
@wateras could you please send a pull-request?