tuanpmt/esp_mqtt #107: Probably crash

mtk3d, 2016-05-28:
Hello, I have a little problem with ESP MQTT. I have built the program, everything works fine, no errors, no problem with flash, but when I plug it with TTL, I saw that:
`mode : sta(18:fe:34:d5:03:85)
add if0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with HomeAutomationNetwork, channel 6
dhcp client start...
ip:192.168.3.100,mask:255.255.255.0,gw:192.168.3.1
pm open,type:2 0`

and i think esp module is crash. Examples from sdk works fine.

Sorry for my english.

sreeisalso, 2016-05-31:
Hi @mtk3d,

There is no crash from your log it is normal.

Thanks and Regards
Sreekanth

mtk3d, 2016-06-04:
But after that I do not see anything more.

Cicero-MF, 2016-06-06:
Are your debug outputs aren't enabled?  See /mqtt/include/debug.h for what you need for INFO() to print.

tuanpmt, 2016-09-09:
Because there are node defined MQTT_DEBUG_ON, check: https://github.com/tuanpmt/esp_mqtt/blob/develop/Makefile#L113 
Now you only need set `FLAVOR=debug` to see all log and `FLAVOR=release` to hide all

closed 2016-09-09 by tuanpmt