tuanpmt/esp_mqtt #114: stops reconnecting when publish'ing while offline

st0ff3r, 2016-06-28:
should system_os_post(MQTT_TASK_PRIO, 0, (os_param_t)client); be called in mqtt_timer regardless of client->connState ?

st0ff3r, 2016-06-30:
apparently fixed by calling MQTT_Connect() from timeout callback. Should that be necessary?

also changed
espconn_disconnect(client->pCon)
to 
client->connState = TCP_RECONNECT_DISCONNECTING
in
mqtt_tcpclient_recv()

documentation says dont call espconn_disconnect() in  any espconn callback

tuanpmt, 2016-09-09:
@st0ff3r i have no problem with calling espconn_disconnect from callback: mqtt_tcpclient_recv. But your right: `documentation says dont call espconn_disconnect() in any espconn callback`
I will prefer to fix this