tuanpmt/esp_mqtt #119: Updated ESP_MQTT

tuanpmt, 2016-09-09:
Hello everyone,
Sorry for Boom mail, I have closed many issues and expected to plan to resolve others. But we have some news:
- esp_mqtt was updated to work with SDK 2.0
- Makefile was cleaned up and easy to use, but need to improve to use with fota firmware 
- Remove esptool submodule (you need to clone from official project)
- Reformat code
- Fix issue call connected callback before disconnect if mqtt username and password not correct 
- Remove `config` module, simplifying the use
- Using `user_config.local.h` to avoid provide Wifi information to github

Thank you,

tuanpmt, 2016-09-09:
@vowstar You really contributed a lot to this project. Please help me noticed changes affect not good

harryd100, 2016-09-10:
Great, just converted one of my existing projects to the latest build and SDK 2.0.0. Works fine.

vowstar, 2016-09-15:
@tuanpmt Ok, I will notice and test these changes on amount of hardware, I'm trying to build a statistical platform to do this.

closed 2016-11-29 by tuanpmt